micropython/micropython #4176: Arm 64-bit port

AntonioND, 2018-09-22:
Is there anyone working on a AArch64 port? I know that you can run a 32-bit binary in a 64-bit system, but there are some restrictions so it would still be nice to have it.

adritium, 2018-09-23:
What would need to be changed for 64-bit port?

AntonioND, 2018-09-23:
Well, it's a completely different instruction set.

drawkula, 2018-09-23:
On a PI3-B (old version without "+") running Devuan2 ...
```
~$ uname -a
Linux pi3-0 4.14.18-v8+ #1 SMP PREEMPT Tue Feb 13 12:54:55 CET 2018 aarch64 GNU/Linux
~$ cat /etc/devuan_version 
ascii
```
... MicroPython builds without problems:
```
/home/wrk/micropython/ports/unix$ ./micropython 
MicroPython v1.9.4-568-g4df194394-dirty on 2018-09-23; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> █
```

AntonioND, 2018-09-23:
Actually, you may still be compiling it for 32-bit. Can you run ```gcc -version```? Your kernel is 64-bit but that doesn't say anything about your toolchain.

drawkula, 2018-09-23:
```
~$ gcc -v |& head -4
Using built-in specs.
COLLECT_GCC=/usr/bin/gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/aarch64-linux-gnu/6/lto-wrapper
Target: aarch64-linux-gnu
```
```
/home/wrk/micropython/ports/unix$ file micropython
micropython: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, for GNU/Linux 3.7.0, BuildID[sha1]=2e234ce1cada0a76c61044295436fec793bbbc79, stripped
```

AntonioND, 2018-09-23:
Interesting. How does micropython work there? Is it just a pure interpreter?

drawkula, 2018-09-23:
I just built it but I never used it.

peterhinch, 2018-09-23:
@AntonioND 
>Interesting. How does micropython work there? Is it just a pure interpreter?

By default code is compiled to bytecode which is interpreted. However it can emit native machine code using **@micropython.native** or **@micropython.viper** decorators, and it also enables writing inline assembler. Clearly these features would need to be ported to a new instruction set.

@dpgeorge is the FP guru but I guess that by default MicroPython assumes a 32 bit hardware FPU. This is just a guess - I haven't studied the code.

AntonioND, 2018-09-23:
I've just compiled the minimal unix target it in my RPi3+ with Fedora and it works out of the box indeed. I guess it's slower than having the machine code compiler but it does the trick for me.

toolmacher, 2018-09-24:
uPy works also on a Renesas RCar-M3 (AArch64), compiled with linaro-toolchain. no big changes necessary. But FPU does not work.

adritium, 2018-09-24:
@AntonioND I'm not saying this is exhaustive but here's the two major **configurations** you have to set up that depend on 32- vs 64-bit.

1. "must be pointer size" - search for this and you'll find a few places where you need to set typedef to the 64-bit signed and unsigned int. In 64-bit, `void*` are 64-bit, right?
2. `MICROPY_OBJ_REPR` - 64-bit precludes representation C.
3. garbage collection - when you're garbage collecting, you need to take care of the registers which are specific to the architecture; see `gc_helper_get_regs_and_sp`

dpgeorge, 2021-03-11:
NLR and GC helper code was added in 2d5cece5ac3fc467055dd3e27378ad497adcee00 and 098ac11bb0cb912909af801015ac07bd2bf01ce7

egbit, 2022-05-19:
Is anyone working on porting py/dynruntime.mk and all it requires to aarch64?
micropython builds easily for aarch64, and I can include C code into the executable.  I'm looking to put C code into .mpy files to be dynamically loaded in micropython on an aarch64 system.  This does not appear supported yet.  For example going into examples/natmod/features0 , edit Makefile to set ARCH = aarch64, type make.

jonnor, 2023-07-30:
@egbit you can try do add something like the following to `py/dynruntime.mk`  - assuming that your host is aarch64.

```
else ifeq ($(ARCH),aarch64)

# aarch64
CROSS = 
CFLAGS += -fno-stack-protector
MICROPY_FLOAT_IMPL ?= double
```

jimmo, 2023-08-03:
@jonnor I get the impression that @egbit's question _is_ about cross-compiling.

To support aarch64 for dynamic native modules there's more work to be done, mostly in `mpy_tool.py` (you can see how all the other supported architectures are implemented there).

I'm going to close this issue because the original issue is solved (there's NLR and GC support for aarch64). There is not yet a native emitter, but as @peterhinch has pointed out this is not necessary. MicroPython just uses the compiler's floating point support (and optionally you can use our libm) so setting `MICROPY_FLOAT_IMPL=double` will use double precision float (either soft or hard depending on compiler flags and target).

closed 2023-08-03 by jimmo