micropython/micropython #4177: renaming a directory into itself looses it and all it's blocks

rhubarbdog, 2018-09-22:
I have a pyboard v1.1 with a SKIPSD file on the flash.
running the following program will create lost blocks on the SD card.  *Only* run it on an SD card you are willing to reformat.
'''
import pyb
import os

os.mount(pyb.SDCard(), '/sd')
os.chdir('/sd')
os.mkdir('test')
print(os.listdir())
print(os.statvfs('/sd'))
os.rename('test','test/new')
print(os.listdir())
print(os.statvfs('/sd'))
os.umount('/sd')
'''

The directory `test` is no longer there
The directory `new` is no where to be seen
The space consumed by directory `test` and all the space consumed by the files and directories within it are lost. The blocks aren't freed up.

dpgeorge, 2018-09-26:
This looks like an issue with the fatfs library/driver, it should probably return an error when trying to do this.  Alternatively the caller of fatfs could check for this case and raise an error (harder to do, better if fatfs does it).

Until this is fixed, just don't rename directories into themselves.

tannewt, 2019-03-13:
Here is logic to detect this:

```c
    // Check to see if we're moving a directory into itself. This occurs when we're moving a
    // directory where the old path is a prefix of the new and the next character is a "/" and thus
    // preserves the original directory name.
    FILINFO fno;
    FRESULT res = f_stat(&self->fatfs, old_path, &fno);
    if (res != FR_OK) {
        mp_raise_OSError(fresult_to_errno_table[res]);
    }
    if ((fno.fattrib & AM_DIR) != 0 &&
            strlen(new_path) > strlen(old_path) &&
            new_path[strlen(old_path)] == '/' &&
            strncmp(old_path, new_path, strlen(old_path)) == 0) {
        mp_raise_OSError(MP_EINVAL);
    }
```

From https://github.com/adafruit/circuitpython/pull/1645