micropython/micropython #4178: [STM32L476] return from pyb.stop() affects working of ADC

rolandvs, 2018-09-22:
It seems that when using the STM32L476 port the use of pyb.stop() and returning from it with rtc.wakeup() do not reenable all used peripherals or clocks (assumption). So even though the program continues from where it left off, some peripherals are affected.

USB is lost which is logical as the connection will be broken... so that is expected.
But see #4022 for the effect it has on the ADC and maybe some more peripherals...

dpgeorge, 2018-09-24:
Thanks for the report.  This should be fixed by 1acf58c08f0f29408522a99cc7ef1d6270c21c09 (that commit also fixes STM32F7xx behaviour as well).

rolandvs, 2018-09-24:
@dpgeorge thanks a lot! Did not check yet, but can we expect the same with the H743? At least another PLL is used for the USB, however did not check if it has similar behaviour yet...

dpgeorge, 2018-10-23:
> Did not check yet, but can we expect the same with the H743

I'm not sure about the H7 for this case.  The H7 still needs a lot of features added, so that will be addressed later.

closed 2018-10-23 by dpgeorge