micropython/micropython #4179: The example propgram, ubluepy_eddystone.py, does not run

rlcase, 2018-09-22:
While testing out my NRF port to the nRF52840-MDK [https://wiki.makerdiary.com/nrf52840-mdk/](https://wiki.makerdiary.com/nrf52840-mdk/), the BLE eddystone example found at
[https://github.com/micropython/micropython/blob/master/ports/nrf/examples/ubluepy_eddystone.py](https://github.com/micropython/micropython/blob/master/ports/nrf/examples/ubluepy_eddystone.py)
issues the following error:
`OSError: Can not fit data into the advertisment packet.`

I am assuming example code should work.

Rich

aykevl, 2018-09-22:
I don't know why it doesn't work but maybe @glennrub knows more.

glennrub, 2018-09-22:
Thanks @rlcase for raising the Issue.

I took a look at it and could reproduce it very quickly. Basically the example is ok, but the driver used wrong value for max advertisement packet size. 
I pushed what i believe is the fix in PR #4180.

rlcase, 2018-09-22:
Wow, that was fast. Thanks.

glennrub, 2018-10-11:
I'm closing this issue as PR#4180 has been merged and should have resolved the bug.

closed 2018-10-11 by glennrub