micropython/micropython #4182: Error building unix port on MacOS

rdlaner, 2018-09-22:
I am attempting to build the unix port of on MacOS, but part way through the build I keep running into this error:

CC ../../lib/axtls/ssl/asn1.c
**error: unknown warning option '-Wno-unused-but-set-variable'; did you mean '-Wno-unused-const-variable'? [-Werror,-Wunknown-warning-option]**
make: *** [build/lib/axtls/ssl/asn1.o] Error 1

When I took a look at the MakeFile, it appears that the build system uses Clang instead of GCC when building on a mac. From what I can tell, the "-Wno-unused-but-set-variable' option is not available with Clang.

Anyway of getting around this? I have GCC installed via homebrew if that is a possibility.

MacOS version: 10.13.6
MicroPython Master branch hash #: 4df194394

adritium, 2018-09-25:
Just remove that warning option

minimalrust, 2018-10-07:
@rdlaner I ran into a similar issue and managed to build by removing the option as suggested.

Try this from outside the micropython directory:

sed -i 's/-Wno-unused-but-set-variable//g' micropython/py/py.mk
cd micropython/ports/unix/
make axtls
make

pfalcon, 2018-10-17:
> build/lib/axtls/ssl/asn1.o

Somehow I missed reference to this file. So, it doesn't look like "ones upon a time somebody tries a strange compiler", this looks like a reaction your @dpgeorge recent axTLS build changes, and thus there's likely something to fix.

Why there're all those warning options for axTLS? It's 3rd-party lib which comes as-is, so the obvious solution is to turn off warning for it at all.

pfalcon, 2018-10-17:
@rdlaner : Suggested to update the title with "... after axTLS build changes".

dpgeorge, 2018-10-30:
Hopefully fixed by 746dbf78d3067f32c8885217d53ff9231dac3772

closed 2018-10-30 by dpgeorge