micropython/micropython #4183: Support for USB_HID Consumer Control devices?

klardotsh, 2018-09-24:
For context in case any of y'all aren't HID folks (I wasn't until recently!), Consumer Mode is an additional (pseudo-)device a USB peripheral can advertise to allow a whole bunch of generally media-related functionalities - notably, the "media keys" of modern keyboards (PlayPause, VolumeUp, VolumeDown, that goofy button that launches your default web browser when you probably meant to hit some other key, etc).

I see some vague references to consumer device reports in the STM32 USB HAL code (I'm testing primarily on a PyBoard, though NRF devices are also in my scope of interest), but nothing notable exposed to Python. I suppose I could use a custom tuple as the `hid` argument to `USB_HID('VSC+HID', hid=...)`, but as far as I understand it, I'd then lose the main keyboard HID device.

I suppose a related question here is how to "be" two HID types at once in MicroPython - let's say for example I wanted to throw a trackpoint and some mouse buttons onto a MicroPython-based keyboard. I'd need to "be" at least two HID types here - keyboard and mouse. Three if I want volume keys on said device - keyboard, mouse, and consumer control.

There's some existing art in this general realm, particularly on NRF chips, over in adafruit/circuitpython land [(see also)](https://github.com/adafruit/Adafruit_CircuitPython_HID/blob/master/adafruit_hid/consumer_control.py), but at this point that project has deviated far enough that I'm nowhere near comfortable enough with C to try to backport their additions.

I'm hoping I'll be told I need to read more closely, and that Consumer Control devices are supported, but from my understanding so far, it looks like there's just basic stubs in C, and no way for me to (easily, out-of-the-box) use them. Thanks folks!

dhalbert, 2018-09-24:
Consumer Control is supported in CircuitPython in both the atmel-samd and nrf ports. Also see https://github.com/adafruit/usb_descriptor/blob/master/adafruit_usb_descriptor/hid.py for some known-working composite USB HID descriptors that combine keyboard, mouse, consumer control, and syscontrol (power) into a single composite USB device that uses only one endpoint.

The C code in CircuitPython doesn't really need to know the details of the different HID devices. It just knows the Report IDs and report lengths for the different kinds of devices. The reports are generated in Python, as you linked to above.

An issue on the STM is that it doesn't support many USB endpoints (just three pairs, I think). The SAMD chips support many more.

klardotsh, 2018-09-24:
So it sounds like, per https://docs.micropython.org/en/latest/pyboard/library/pyb.html#pyb.usb_mode...

>If enabling HID mode, you may also specify the HID details by passing the hid keyword parameter. It takes a tuple of (subclass, protocol, max packet length, polling interval, report descriptor).

...that I could copy the giant byte array of the quad-descriptor from the `hid.py` you linked, pass it as the final member of the `hid=` tuple, update my application code to send the significantly longer `bytearray`s as reports, and in theory get this working while not sabotaging the other uses of my USB port (read: serial for REPL), even on STM? Or am I misunderstanding.

I'm fine with sacrificing the UART REPL if I have to, but if I'm understanding you correctly, I should have space (assuming three pairs can run over USB, I'd have REPL, KB+Mouse, Consumer+syscontrol)?

Sorry for sitting squarely in that weird category of knowing what questions to ask but not fully how to understand the answers yet! Micro/Circuit Python are whole new worlds to me (though CPython and to some extent basic C are not), but thanks for your help here and in AdaFruit Discord!

dhalbert, 2018-09-24:
It'd be KB+Mouse+Consumer+SysControl all on one endpoint. But the reports have to include the report ID to specify the device.

I'm not familiar with how HID works on the STM32 port, so I'm not going to predict the success of this :)

klardotsh, 2018-09-30:
Well, my application code needs some serious tweaking to send the new reports correctly (maybe I'll just try to port the AdaFruit HID abstractions to MicroPython/my codebase, since this codebase _also_ targets CircuitPython anyway), but, great news:

```
[ 2993.075708] usb 1-4: new full-speed USB device number 33 using xhci_hcd
[ 2993.217032] usb 1-4: New USB device found, idVendor=f055, idProduct=9801, bcdDevice= 2.00
[ 2993.217039] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 2993.217043] usb 1-4: Product: Pyboard Virtual Comm Port in FS Mode
[ 2993.217047] usb 1-4: Manufacturer: MicroPython
[ 2993.217050] usb 1-4: SerialNumber: 365332743036
[ 2993.223509] input: MicroPython Pyboard Virtual Comm Port in FS Mode Keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:F055:9801.001E/input/input54
[ 2993.279545] input: MicroPython Pyboard Virtual Comm Port in FS Mode Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:F055:9801.001E/input/input55
[ 2993.280014] input: MicroPython Pyboard Virtual Comm Port in FS Mode Consumer Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:F055:9801.001E/input/input56
[ 2993.280285] input: MicroPython Pyboard Virtual Comm Port in FS Mode System Control as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:F055:9801.001E/input/input57
[ 2993.280631] hid-generic 0003:F055:9801.001E: input,hidraw11: USB HID v1.11 Keyboard [MicroPython Pyboard Virtual Comm Port in FS Mode] on usb-0000:00:14.0-4/input0
[ 2993.280856] cdc_acm 1-4:1.1: ttyACM0: USB ACM device
```

That's all from this `boot.py` on the PyBoard:

```python
import pyb

report_structure = bytes([
    0x05, 0x01,     # Usage Page (Generic Desktop),
    0x09, 0x02,     # Usage (Mouse),
    0xA1, 0x01,     #  Collection (Application),
    0x09, 0x01,     #   Usage (Pointer),
    0xA1, 0x00,     #  Collection (Physical),
    0x05, 0x09,     #     Usage Page (Buttons),
    0x19, 0x01,     #     Usage Minimum (01),
    0x29, 0x03,     #     Usage Maximum (03),
    0x15, 0x00,     #     Logical Minimum (0),
    0x25, 0x01,     #     Logical Maximum (1),
    0x75, 0x01,     #     Report Size (1),
    0x95, 0x03,     #     Report Count (3),
    0x81, 0x02,     #     Input (Data, Variable, Absolute)
    0x75, 0x05,     #     Report Size (5),
    0x95, 0x01,     #     Report Count (1),
    0x81, 0x01,     #     Input (Constant),
    0x05, 0x01,     #     Usage Page (Generic Desktop),
    0x09, 0x30,     #     Usage (X),
    0x09, 0x31,     #     Usage (Y),
    0x09, 0x38,     #     Usage (Scroll),
    0x15, 0x81,     #     Logical Minimum (-127),
    0x25, 0x7F,     #     Logical Maximum (127),
    0x75, 0x08,     #     Report Size (8),
    0x95, 0x03,     #     Report Count (3),
    0x81, 0x06,     #     Input (Data, Variable, Relative)
    0xC0,           #  End Collection,
    0xC0,           # End Collection
])

report_structure = bytes([
    # Regular keyboard
    0x05, 0x01,  # Usage Page (Generic Desktop)
    0x09, 0x06,  # Usage (Keyboard)
    0xA1, 0x01,  # Collection (Application)
    0x85, 1,     #   Report ID (1)
    0x05, 0x07,  #   Usage Page (Keyboard)
    0x19, 224,   #   Usage Minimum (224)
    0x29, 231,   #   Usage Maximum (231)
    0x15, 0x00,  #   Logical Minimum (0)
    0x25, 0x01,  #   Logical Maximum (1)
    0x75, 0x01,  #   Report Size (1)
    0x95, 0x08,  #   Report Count (8)
    0x81, 0x02,  #   Input (Data, Variable, Absolute)
    0x81, 0x01,  #   Input (Constant)
    0x19, 0x00,  #   Usage Minimum (0)
    0x29, 101,   #   Usage Maximum (101)
    0x15, 0x00,  #   Logical Minimum (0)
[... 61 lines cut ...]
    0xC0,              # End Collection
    # Power controls
    0x05, 0x01,        # Usage Page (Generic Desktop Ctrls)
    0x09, 0x80,        # Usage (Sys Control)
    0xA1, 0x01,        # Collection (Application)
    0x85, 4,           # Report ID (n)
    0x75, 0x02,        #   Report Size (2)
    0x95, 0x01,        #   Report Count (1)
    0x15, 0x01,        #   Logical Minimum (1)
    0x25, 0x03,        #   Logical Maximum (3)
    0x09, 0x82,        #   Usage (Sys Sleep)
    0x09, 0x81,        #   Usage (Sys Power Down)
    0x09, 0x83,        #   Usage (Sys Wake Up)
    0x81, 0x60,        #   Input (Data,Array,Abs,No Wrap,Linear,No Preferred State,Null State)
    0x75, 0x06,        #   Report Size (6)
    0x81, 0x03,        #   Input (Const,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
    0xC0,              # End Collection
])

existing_keyboard = list(pyb.hid_keyboard)
existing_keyboard[-1] = report_structure
hid_super_keyboard = tuple(existing_keyboard)

pyb.usb_mode('VCP+HID', hid=hid_super_keyboard)  # act as a serial device and a mouse
```

klardotsh, 2018-09-30:
I also just tested muting and unmuting my system volume successfully:

```python
MicroPython 460954734 on 2018-09-29; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import time
>>> from pyb import USB_HID
>>> hid=USB_HID()
>>> hid.send((3,0x00E2)); time.sleep(0.01); hid.send((3,0))
2
2
>>> hid.send((3,0x00E2)); time.sleep(0.01); hid.send((3,0))
2
2
```

What I can't figure out is where on earth my keyboard went - sending an eight-part tuple with report IDs of 0, 1, or 2 all seem to fail, but this may be PEBKAC. Calling it a day on this - looks promising so far!

closed 2018-10-03 by klardotsh