micropython/micropython #4198: Convert Arduino Code to Micropython

dpnkrg, 2018-09-29:
### how to convert Arduino code to Micropython ?

for example i want to convert this arduino code to micropython.

```
// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}
```

peterhinch, 2018-09-29:
Please raise support queries in [the forum](https://forum.micropython.org/). This is the MicroPython bug tracker.

dpnkrg, 2018-09-29:
okay, thanks.
closing this.

closed 2018-09-29 by dpnkrg