micropython/micropython #4200: protobuf

UBTEDU, 2018-09-30:
Support the protobuf?

dpgeorge, 2018-10-01:
If you mean Google Protocol Buffers, then, no, they are not currently supported.  But there is a nomal Python library for protobuf and you could try running that as-is under MicroPython.

goatchurchprime, 2018-10-15:
This would probably go into a separate issue when it's completed, but I've watched @mcqn spend the last week porting https://github.com/signalapp/libsignal-protocol-c to the Arduino/ESP32 platform for one of his projects that needs really secure messaging between two IoT devices over MQTT.  

So I searched the MicroPython issues for "protocol buffers" because it appears to be one of the dependencies for signal.  

Just putting this out there as a future addition to the potential options for making strong security easily available to all.

jpochyla, 2018-10-15:
Protobuf is really easy to encode/decode, we use this async codec (GPLv3): https://github.com/trezor/trezor-core/blob/master/src/protobuf.py

Should be quite easy to remove the async support and make it read/write from BytesIO, for example. So feel free.

UBTEDU, 2018-12-26:
Can  support protobuf? like nanopb can support arduino and stm32
https://github.com/nanopb/nanopb

wgaylord, 2019-06-18:
I am interested in this as the protobuf  python library takes up 2.4 MB of space. (Haven't tried to minify it at all yet.)

dogtopus, 2020-01-24:
I just tested my own small Protobuf library under MicroPython and to my suprise it seems to work well with only minor hacks. The library is only about 8KB in size after compiled with `mpy-cross`. It can be found at https://github.com/dogtopus/minipb if anyone is interested.

peterhinch, 2020-01-24:
@dogtopus This looks like a promising "micro" implementation.

wgaylord, 2020-01-24:
Thanks! This will be really useful to me.

MrSurly, 2020-01-24:
Just so I understand we  **are not** talking about: Using something akin to `protoc` to _generate_ code to read/write a _particular_ message type.

We **are** talking about generic code that parses a `.proto` file at runtime to read/write messages.

dogtopus, 2020-01-24:
@MrSurly Actually I skipped the proto file entirely and used a format string similar to what struct uses for packing/unpacking. The code then use this format string to serialize from/deserialize to tuples. In addition it also accepts a list of fields and types similar to the one used in ctypes `Structure._fields_`. In this mode it serializes from/deserializes to dictionaries. It works currently but I still need to write more documentations for it.

Generating schema and classes from proto files are theoretically possible at this point but I haven't implemented it yet.

MrSurly, 2020-01-24:
Weirdly, I'm currently working on a C deserializer that works directly with Python `struct` format strings.  It generate very basic C code that can take this format, and a binary buffer, and fill a C struct. 

Of course python structs are a bit limited since they don't support variable length strings.

dogtopus, 2020-01-24:
Well if C needs to be involved in some way, nanopb might be an option as well since it's also small footprint and should play nicely with Python with some extra bindings.

MrSurly, 2020-01-24:
Sorry, I wasn't clear -- the target isn't Python.  The idea is being able to generate binary structs in Python using `struct`, but deserializing in C. Specifically C90 w/o any heap functions.

tve, 2020-01-25:
Dumb question: this is a very interesting discussion but I'm wondering whether anyone is actually asking that something be added to this repo. Wouldn't a protobuf implementation live in its own repo completely independent of the MP repo?

peterhinch, 2020-01-25:
The OP appeared to ask if protobuf was officially supported, and received the answer that it was not. 

A discussion of third party libraries ensued, whereupon @dogtopus pointed us at his solution. This works well on MicroPython baremetal targets. Nobody asked for it to be incorporated in official builds. It's a great user-maintained resource for MicroPython projects.

So the answer to your question is "yes". In my opinion, of course.

peterhinch, 2020-01-25:
To add to the above the solution of @dogtopus is licensed under LGPLv3+ which would rule it out. Alas this license would also rule it out from most commercial MicroPython and CPython applications.

@dogtopus Why did you opt for LGPL?

dogtopus, 2020-01-25:
>Why did you opt for LGPL?

{L,}GPL is more or less just a safe default if I'm unsure what license to use. I do notice that Protobuf uses BSD and I'm okay with switching to a more permissive license like BSD or MIT if requested.

dogtopus, 2020-01-26:
Also AFAIK LGPL did not forbid commercial use as long as the library (in this case minipb) is sufficiently independent from the rest of the user code (like no copy-pasting) and any modifications to the source code of the library is provided. So commercial projects would be fine as long as they figure out a way to put a link to the minipb github repo somewhere (even including sending the link to anyone who asked for it) or publish their modified code if there's any modification so we can benefit from it.

peterhinch, 2020-01-26:
IANAL and this probably isn't the place for an extended discussion but it seems to be widely accepted that including LGPL code in any project affects the entire project. I have read a number of times that commercial users refuse to include GPL code in a closed-source project.

While I'm unsure of the legal details I do know that @dpgeorge won't allow GPL code anywhere near official MIT licensed MicroPython. Contributors must guarantee that nothing was copied from, or based on, GPL licensed code.

For wide acceptance I would recommend a permissive licence unless there is a good reason for using GPL.

dogtopus, 2020-01-26:
>While I'm unsure of the legal details I do know that @dpgeorge won't allow GPL code anywhere near official MIT licensed MicroPython. Contributors must guarantee that nothing was copied from, or based on, GPL licensed code.

>(like no copy-pasting)

Direct integration like **copy-pasting** GPL or LGPL code to non-GPL project affects the whole project. However **calling** an **LGPL** library inside non-GPL project OTOH is different.

However I do agree that on embedded project things can get a bit tricky (like static linking vs dynamic no longer makes sense. That's one more reason for permissive embedded C projects like MicroPython to stay GPL-free) but seems that MicroPython already did a great job for isolating library code from main user code (as in users can upload mpy files to PYBFLASH and invoke in their code by importing).

gretel, 2020-12-13:
`nanopb` is https://github.com/nanopb/nanopb/blob/master/LICENSE.txt not any `*GPL*`

jonnor, 2024-09-19:
protobuf support will likely always be in an external module. There is a seemingly good one linked above. So I think we can close this issue now. Future discussions on such topics (that can be handled as third party module), should probably happen in the Discussions: https://github.com/orgs/micropython/discussions

closed 2024-09-19 by jonnor