micropython/micropython #4202: HID on Windows

kdb424, 2018-10-01:
Curious about the status of HID on windows. I am currently running (CDC+HID) on my pyboard, and Windows 10 is not getting keystrokes. Having absolutely no issues on Linux. Is this a problem with my mostly fresh Windows 10 install, or is this something that I'm doing incorrectly. All keystrokes going through properly on Linux.

klardotsh, 2018-10-01:
Just confirming this is the case on Windows 7 as well - Windows throws a fit about the COM lane not having a driver and seemingly fails the entire USB port, despite seeing a HID Keyboard with Microsoft's generic HID driver and the system claiming that device lane "works" (it does not).

klardotsh, 2018-10-01:
I'm using `VCP+HID` and have tried both the built-in PID/VID as well as the PID/VID of one of my other keyboards laying around (`V: ceeb P:1256`, which is a Keebio Iris keyboard with Keyboard, Mouse, Consumer, SysControl composite). Windows gets no input in either case,  but the device works great in Linux.

I've also gone into Device Manager and disabled the COM device specifically that shows up... still no HID input, despite (as usual) Windows telling me the HID keyboard and mouse are "working" on this lane.

My `boot.py` is a very slightly modified version of what I shared [here](https://github.com/micropython/micropython/issues/4183#issuecomment-425683279), for what it's worth.

The only difference between @kdb424 and I's setup (we're working on the same project) is that he's testing on bare metal Win10 x64, I'm testing through VirtualBox (Windows 7 x32 and 10 x64). Same symptoms, though.

klardotsh, 2018-10-01:
Wondering if I need to `recv()` some data and/or send some initialization packet on Windows specifically? I basically never touch Windows so this is all a mystery to me.

klardotsh, 2018-10-09:
Any chance of a bump on this? I still haven't been able to get the HID keyboard Windows sees to actually do anything, on Windows only, specifically.

dpgeorge, 2018-10-11:
I tested USB HID support with a PYBv1.0 and native (not virtual) Windows 10 v1803, and it works ok for me.  I tested both mouse and keyboard and they both work.

For mouse test:
```python
pyb.usb_mode('VCP+HID') # in boot.py

h = pyb.USB_HID()
h.send(b'\x00\x10\x00\x00') # makes mouse move to the right by 16 pixels
```

For keyboard test:
```python
pyb.usb_mod('VCP+HID', hid=pyb.hid_keyboard) # in boot.py

import time
h = pyb.USB_HID()
def kb_press(x):
    b = bytearray(8)
    b[2] = x # key press
    h.send(b)
    time.sleep_ms(25)
    b[2] = 0 # key release
    h.send(b)
kb_press(4) # press "a" once
```

I can successfully use the serial COM port while running these tests, so VCP and HID do work together at the same time in my case.

It might be worth trying a fresh Windows machine that didn't yet see any pyboard before, and see if VCP+HID works on it.  Because in that case it must install the drivers from scratch.  Under Windows 10 it should do it automatically.

---

If nothing works then there is the option to add to the firmware a HID-only USB mode, like `pyb.usb_mode('HID')`.  That would most likely work because it's just a simple HID device, no interface association descriptor is needed.

klardotsh, 2018-10-11:
Might this have something to do with my use of a giant multi-descriptor to do Keyboard,Mouse,Consumer,SysControl all on one huge composite device? The code in question is in these [two](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/micropython/pyb_hid.py) [files](https://github.com/KMKfw/kmk_firmware/blob/master/kmk/common/abstract/hid.py). There's some notes in there about hackarounds I've had to do for various reasons, but tl;dr of how I got composite devices working on Linux was with a bytearray where the first byte determines the device sending a report (`x03` for Consumer, for example)

dpgeorge, 2018-10-11:
> Might this have something to do with my use of a giant multi-descriptor to do Keyboard,Mouse,Consumer,SysControl all on one huge composite device?

Very likely.  Did you try the standard mouse or keyboard-only descriptors that are built in to the firmware (eg the tests I describe above)?  If they work on your Windows set up then it is highly likely to be an issue with the descriptors.  I would then recommend to start with just a keyboard with your own code, then add a mouse, then a consumer, etc, and see where it stops working.

BTW, USB_HID does support ioctl to poll if it is ready to write to again.

dhalbert, 2018-10-11:
The large descriptor described was above works fine on Windows on SAMD processors, assuming it was not changed (I wrote and debugged it on Linux, Windows 7 and 10, and Mac). So it may be a combination of the descriptor and how the USB stack is implemented on the STM processors.

Another thing to note is that the processor on the pyboard has a limited number of endpoints (3 pairs, I think). The SAMD's have 8 pairs. I don't think that's the problem here, since it seems to work for you on Linux, but it's worth watching out for.

jonnor, 2024-09-19:
Problem does not seem to have enough information to reproduce. And some useful pointers were given.

closed 2024-09-19 by jonnor