micropython/micropython #4203: cc3200 upip error

m2030, 2018-10-01:
recently i run micropython  on my simplelink CC3200 LaunchPad as steps : 
[(http://processors.wiki.ti.com/index.php/User:Dbartlett/MicroPython_CC3200)]
but i cant find upip and i have not find any way to install it on my board 
has any one find way to install?
  the error from repl as [following]
` import upip
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: module not found`

m2030, 2018-10-02:
Thanks Mr.Peter it is done

pfalcon, 2018-10-02:
The simple answer to this question is that upip is shipped as builtin only with unix, esp8266 (and probably esp32) ports. For all other ports, upip should be installed manually to the filesystem first.

m2030, 2018-10-02:
Thanks for your concern
I wonder if I installed it to lib directory or other directory(sys,cert)?

jonnor, 2024-09-19:
upip has been replaced by mip, see https://docs.micropython.org/en/latest/reference/packages.html

closed 2024-09-19 by jonnor