micropython/micropython #4205: STM micropython.heap_unlock() causes memory error

peterhinch, 2018-10-03:
From [this forum thread](https://forum.micropython.org/viewtopic.php?f=6&t=5329).
```python
MicroPython v1.9.4-623-g34af10d on 2018-10-03; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> 
>>> import micropython
>>> micropython.heap_lock()
>>> micropython.heap_unlock()
MemoryError: 
>>> 
```
The error also occurs if you issue
```python
MicroPython v1.9.4-623-g34af10d on 2018-10-03; PYBv1.1 with STM32F405RG
Type "help()" for more information.
>>> import micropython
>>> micropython.heap_unlock()
>>> micropython.heap_unlock()
MemoryError: 
>>> 
```

pfalcon, 2018-10-03:
Sure, that's how it works, and how else could it be? Just inputting a line from console requires allocation, which heap_lock() precludes. It's not intended or suitable for interactive use.

peterhinch, 2018-10-03:
Thank you. That accounts for the first case but I'm still puzzled by the second.

pfalcon, 2018-10-03:
Apparently no underflow checking. You can check the source to see if that's true. It behaves the same way on unix port too, which may be more comfortable to experiment with.

robert-hh, 2018-10-03:
Actually: in py/gc.c, function gc_unlock() as suitable change my be:
```
void gc_unlock(void) {
    GC_ENTER();
    if (MP_STATE_MEM(gc_lock_depth) > 0) {
        MP_STATE_MEM(gc_lock_depth)--;
    }
    GC_EXIT();
}
```

adritium, 2018-10-03:
So that should be fixed?

https://github.com/micropython/micropython/blob/34af10d2ef545ecaf2c1824dac679af39d683e98/py/gc.c#L181-L185

peterhinch, 2018-10-04:
It does fix it. I've raised https://github.com/micropython/micropython/pull/4209.