micropython/micropython #4207: [NUCLEO_H743ZI] changing CPU frequency does not work as expected

rolandvs, 2018-10-03:
With the latest release 1.9.4 tested but also in 1.9.3 (just to be sure) I use `machine.freq()` to set the CPU frequency to a low frequency. Setting it to something other than 400MHz (machine.freq(200000000)) the REPL interface via the ST-LINK (real UART, no USB) will not work (garbled, not responsive).

Using REPL via USB will still work. However something like setting the frequency to 10_000_000 (10MHz) of the CPU will give no reply on the UART (expected), but the USB is dead also. After a while RED/BLUE led turn on (Fault) and the USB REPL was just able to say some last words "
can'can'can't change freq". After this only the RED led is on.
 
This could be considered a feature as the clock for USB is derived from another PLL and the clock for the used UART probably uses the same as for the CPU (need to check) and changes accordingly. 

It makes it a bit hard to throttle the CPU speed as other peripherals will also suffer from the change.

adritium, 2018-10-03:
If changing CPU frequency has side-effects then so be it. Unless the documentation for `machine.freq()`  promises no side-effects, there's not a real issue here.

The real issue is in your title: `as expected`.

dpgeorge, 2018-10-04:
There are a few intertwined issues here: 1) whether changing the frequency on the H7 works at all; 2) improving the frequency selection on the H7 (since it has a huge clock tree with lots of options it would be nice to be able to just scale the CPU frequency while keeping peripherals constant); 3) having a working UART REPL after changing the frequency.

For (3), I do get tripped up on this point a lot, and usually make a helper function that changes the CPU frequency and reinits the UART baudrate at the same time.  But maybe it would be a good idea to do this automatically: for any UART that is connected to the REPL, reinit its baudrate when changing the CPU frequency.

I don't think it makes sense to try and automatically adjust all peripherals (I2C, SPI, UART, ADC, etc) when the CPU frequency changes because that's just too difficult and probably they can't even be kept at the same peripheral frequency for any given CPU frequency (eg SPI only has a very limited set of prescalers).

rolandvs, 2018-10-04:
Damien,

Thanks for your response.

I agree that the H7 clock verification is one thing to do and it would be nice to have REPL working under all possible conditions. It is an enormous versatile piece of silicon :-)

For now I will wrap the machine.freq() and add some additional code to reinit some peripherals. If that is working it may be interesting to integrate such mechanism. 

Maybe while fiddling with MX-CUBE clock for the various clock settings, maybe moving around some clocks will solve part of it (as with USB).