micropython/micropython #4210: stm32: if plugging in USB during pyb.delay or pyb.udelay, USB breaks

ryannathans, 2018-10-04:
main.py:
```
while True:
	print('hello')
	pyb.delay(500)  # pyb.udelay(500000) has the same behavior

```

Running the above snippet on a v1.9.4 PYBLITE V1.0, if USB is unplugged and plugged back in, the USB breaks.

Steps to reproduce:
1. plug into USB (device powers up)
2. apply external power supply
3. connect to REPL, observe 'hello' being printed
4. unplug USB
5. plug USB back in
6. connect to REPL, observe nothing being printed anymore (but connection in PuTTY stays open and filesystem is still accessible)

Oddly, **if the pyb.delay is removed, the device works exactly as expected** - no bugs.

Perhaps if USB is connected during a pyb.delay it isn't processed properly?

adritium, 2018-10-04:
@ryannathans 

Do you have a debugger to look into the C code?

ryannathans, 2018-10-05:
@adritium have a debugger but haven't previously been successful getting it attached with micropython. Probably some Windows quirk

ryannathans, 2018-10-05:
Reading the code, it seems like the HAL is a bit.. sucky...

"is connected" appears to only be set to false when a exit control command is received - which won't ever occur if USB is unexpectedly unplugged. Additionally, there appears to be no (obvious) way to reload/restart the CDC driver. I still don't know why it actually dies..

dpgeorge, 2018-10-05:
> Additionally, there appears to be no (obvious) way to reload/restart the CDC driver.

You can restart the USB on the pyboard manually using:
```python
pyb.usb_mode(None)
pyb.usb_mode('VCP+MSC')
```

ryannathans, 2018-10-05:
@dpgeorge now I just need to find a way to detect the problem has occurred from the device end, which might be difficult considering isconnected continues returning True

dpgeorge, 2018-10-05:
> I just need to find a way to detect the problem has occurred from the device end

You should be able to check a bit in the USB peripheral to see if it is connected or not.  Eg in OTG_FS_GOTCTL the BSVLD and CIDSTS should work.  Try something like this (valid for F4 with USB FS):
```python
import stm
usb_status = (stm.mem32[0x50000000] >> 16) & 0xf
```
the status value should have bits 0 and 3 set if it's connected in device mode.  They should be 0 when disconnected.

chrismas9, 2018-10-05:
@ryannathans 
> Probably some Windows quirk

I use Atollic True Studio. It is prepackaged Eclipse, gcc any gdb with good support for JTAG / SWD debuggers under Windows including Jlink and STlink. It is now free from ST Microelectronics.

ryannathans, 2018-10-11:
@dpgeorge work around tested and functions, except because we have USB FS disabled we're detecting the timeout of vcp.send which starts to occur a short while after the event. Not sure about the root cause of the bug though

dpgeorge, 2018-10-11:
> work around tested and functions,

Could you explain in a bit more detail how you implemented the workaround?

> except because we have USB FS disabled we're detecting the timeout of vcp.send which starts to occur a short while after the event

If you mean that vcp.send times out after the USB is unplugged, then I guess that would be expected because there is no longer a serial connection.  If the USB link is down, vcp.send shouldn't be called.

ryannathans, 2018-10-11:
@dpgeorge 

- Our USB mode is VCP only
- Vcp.isconnected remains True indefinitely after USB is unplugged
- A while after usb is unplugged (not instantly), vcp.send starts to time out, I'm not sure what this duration is
- using a timer, our normal vcp.send() duration is measured (inside a "if vcp.isconnected():" which returns true if USB is unexpectedly unplugged)
- if we detect a timeout, usb mode is reset as you suggested by setting to None and then back to VCP

I agree vcp.send shouldn't be called if USB is down, but there doesn't appear to be any clear way of detecting this in usb mode VCP only

dpgeorge, 2018-10-15:
In 53ccbe6cecb0988070207ac21ec628d364489659 I fixed USB VCP so that it handles the disconnect IRQ: when the USB device is disconnected (or the serial port is closed on the host) then `vcp.isconnected()` will return False again.

jonnor, 2024-09-29:
Is this issue still present in latest MicroPython, version 1.23 or later? Many fixes have been made over the last 6 years.

jonnor, 2025-03-03:
Presumably fixed.