micropython/micropython #4212: Rounding error when printing the number 460.0

Ecco, 2018-10-04:
MicroPython doesn't display the number `460.0` like CPython.

CPython
```
>>> 460.0
460.0
```

MicroPython
```
>>> 460.0
460.0000000000001
```

dpgeorge, 2018-10-04:
There are lots of these cases, it's not just 460.0.  The parsing of the float is fine, the problem lies in the float formatting and rounding of the output.  See #1450 for a related (exactly the same?) issue.

As a workaround, if you need well behaved printing of floats then it's suggested to use a specifier like `"{:.6f}"`.

dpgeorge, 2018-10-14:
> This is not a rounding problem.

I think it is a rounding problem, but it's the rounding (or lack of it) done in the float formatting/printing.  When formatting a floating number for printing it uses float arithmetic which can lead to some types of rounding errors.

yur-spiridonov, 2019-11-04:
The number 460.0000000000001 in the internal extended format will be represented by the binary code 111001100.00000000000000000000000000000000000000000001 | 111. If this number is rounded to the low bit, then we get 111001100.00000000000000000000000000000000000000000010 =
4.60000000000000113686837721616E2. If 16 significant decimal digits are displayed in a line, then we get 4.600000000000001.
If rounding is performed without Guard Digits, truncating all bits after the 53rd, then we will have 111001100.00000000000000000000000000000000000000000001 = 460.00000000000005684341886080801486968994140625. If we print 16 significant digits, we get the number 460.0000000000000. Maybe MicroPython and CPython can round binary numbers differently.

goatchurchprime, 2020-03-13:
The function [mp_format_float()](https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/py/formatfloat.c#L114) appears to implement the float to string conversion when `(MICROPY_FLOAT_IMPL != MICROPY_FLOAT_IMPL_NONE)` (the options are MICROPY_FLOAT_IMPL_FLOAT or MICROPY_FLOAT_IMPL_DOUBLE)

The ESP8266 settings define [MICROPY_FLOAT_IMPL = MICROPY_FLOAT_IMPL_FLOAT](https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/ports/esp8266/mpconfigport.h#L92)

The ESP32 settings also define [MICROPY_FLOAT_IMPL = MICROPY_FLOAT_IMPL_FLOAT](https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/ports/esp32/mpconfigport.h#L47)

So, it looks like both the ESP8266 and the ESP32 are using exactly the same source code for `mp_format_float()`, but getting different results.  

So is the answer to be found in this function?  If so I could take a closer look (given that I've just managed to compile for ESP32 in order to gain early access to the I2S interface).  

This is a bug -- there are no rounding errors in floating points to the left of the decimal points.  Integers are perfectly represented when they fit into the mantissa.

dpgeorge, 2020-03-13:
> So, it looks like both the ESP8266 and the ESP32 are using exactly the same source code for `mp_format_float()`, but getting different results.

They use different object representations: esp8266 uses `MICROPY_OBJ_REPR_C`, while esp32 uses `MICROPY_OBJ_REPR_A`.  The former has 30-bit FP numbers because they are boxed into a 32-bit word.

goatchurchprime, 2020-03-13:
Carefully printout out the bitpatterns of the value of `float f` shows that we have the correct number in 
[line 261](https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/py/formatfloat.c#L261), but then we divide out by powers of 2 to make the value between 1 and 10. 

This explains where the inaccuracy is coming from, because 99.0 is converted to 9.9, which then becomes 0.9, then 9.0001, and so the problem arises.  It's nothing to do with the Micropython internal floating point packing.

(I did not know we were not getting the full quid for the floating point numbers.  This could have implications when trying to do signal processing, and not getting the same answers.  There are similar problems with intel processors where the processor floating point registers were 80bits, so you could get different answers from your arithmetic equations depending exactly how the compiler allocated the registers.)

I've been trying to find how this is done elsewhere.  In CPython:
*  [char* PyOS_double_to_string(double val,...)](https://github.com/python/cpython/blob/e9e7d284c434768333fdfb53a3663eae74cb995a/Python/pystrtod.c#L796)
* calls [static char* _PyOS_ascii_formatd(..., double d,...)](https://github.com/python/cpython/blob/e9e7d284c434768333fdfb53a3663eae74cb995a/Python/pystrtod.c#L715)
*  which, "to do the hard work", calls [PyOS_snprintf(buffer, buf_size, format, d)](https://github.com/python/cpython/blob/e42b705188271da108de42b55d9344642170aa2b/Python/mysnprintf.c#L40) 
* which calls [int PyOS_vsnprintf(char *str, size_t size, const char  *format, va_list va)](https://github.com/python/cpython/blob/e42b705188271da108de42b55d9344642170aa2b/Python/mysnprintf.c#L53)
* which calls out to `vsnprintf` in the stdlib.c library.

I think I've found a several thousand line implementation of this last function in a c99 codebase here:
https://github.com/weiss/c99-snprintf/blob/master/snprintf.c#L1260

This appears to use a simple convert() function to convert the integer and fractional parts separately.

Was there a reason not to rely on `vsnprintf()` for this formatting as they do in CPython?  (It's an odd decision, as you'll get different answers depending on the compiler.)  A fix for this problem would involve separating out the integer component of the floating point number first, and not dividing it out by a factor and multiplying it up again.

goatchurchprime, 2020-03-13:
The numbers are all the same on the ESP8266.  The only difference is that the 99.0 is rendered to 4 decimal places instead of 5, and therefore this bad digit falls off the bottom.

```Python
>>> print(9000160.0)
9000159.0
```

The code in question that would need to be looked at is:
```C
static const float neg_pow[] = { 1e-32, 1e-16, 1e-8, 1e-4, 1e-2, 1e-1 };
static const float g_pos_pow[] = { 1e32, 1e16, 1e8, 1e4, 1e2, 1e1 };
int e, e1;
for (e = 0, e1 = 32; e1; e1 >>= 1, pos_pow++, neg_pow++) {
    if (*pos_pow <= f) {
        e += e1;
        f *= *neg_pow;
    }
}
if ((int)f >= 10) { e += 1; f *= FPCONST(0.1); }

int dec = e;
for (int i = 0; i < num_digits; ++i, --dec) {
    int32_t d = (int32_t)f;
    if (d < 0) 
        *s++ = '0';
    } else {
        *s++ = '0' + d;
    }
    if (dec == 0 && prec > 0) {
        *s++ = '.';
    }
    f -= (float)d;
    f *= (float)(10.0);
}
```
This is a very compact and efficient implementation.  I don't immediately see any quick fix.  The C99 code uses an `unsigned long int` to hold all the integer part to as many digits as necessary before converting them to the ascii string of decimal digits, but we don't have that.  We'd need some kind of extra register that effectively cancelled the 10 and 0.1 factors before they were multiplied out and lost the accuracy.

dpgeorge, 2020-03-20:
> Was there a reason not to rely on `vsnprintf()` for this formatting as they do in CPython?

Because most bare-metal ports will not have this function available.  In fact, MicroPython can supply its own version of `vsnprintf()` but that delegates precisely to `mp_format_float()` to format the floats.

dlech, 2020-03-28:
This is more of a problem now that we are using `black` to format code since it doesn't allow floating point numbers with only a trailing `.` it always adds a `0` after the `.`.

The error comes from `mp_parse_num_decimal()` where the number is multiplied by 10 for each trailing 0 before being divided down back to the original number later.

https://github.com/micropython/micropython/blob/f6375ac3ebac28656a0a757952d32c265b1ba7aa/py/parsenum.c#L254

Would it be too much of a performance hit to check that all digits after `.` are 0? Or perhaps just test for the case where there is only one digit after `.` and that it is `0`?

Kristine1975, 2021-11-19:
Accurately printing floating-point numbers is non-trivial. There are various algorithms with different advantages/drawbacks, such as Grisu2, Grisu3, Schubfach, Ryu, Dragonbox... Apple has an [Apache-licensed implementation](https://github.com/apple/swift/blob/main/stdlib/public/runtime/SwiftDtoa.cpp), but it's probably too large for MicroPython ([this pull request has some size comparisons](https://github.com/apple/swift/pull/35299]).

dpgeorge, 2022-06-24:
PR #5766 did fix formatting this particular number (460.0), but broke other numbers.  It may be worth investigating that PR in more detail to see if it can be improved.

bwhitman, 2022-07-07:
Thank you @dpwe!! Confirmed this fixes the issue on esp32 with `MICROPY_FLOAT_IMPL_FLOAT`.

Before:

```python
>>> '%.4f' % 572
'572.0000'
>>> '%.5f' % 572
'571.99998'
```

After:

```python
>>> '%.4f' % 572
'572.0000'
>>> '%.5f' % 572
'572.00000'
```

👍

dpgeorge, 2022-07-26:
Fixed by f9cbe6bc47dd4f5b8e85178caecd6f0de22b4c34