micropython/micropython #4214: ESP32 port missing "hard" kwarg in pin.irq()

jedie, 2018-10-05:
The ESP8266 has `hard` keyword argument in `pin.irq()` function. This is missing in ESP32 port.

See also: https://forum.micropython.org/viewtopic.php?p=30748#p30745

peterhinch, 2018-10-05:
It's also undocumented. Is support intended to be cross-platform, or is the option deprecated? I'm happy to submit a docs PR for **machine** if the option is to stay.

pumelo, 2019-11-12:
I was wondering about this too. Diging into the code I found the reason:

https://github.com/micropython/micropython/blob/673db939b5c1f42c9411f684343d39811b7a16b2/ports/esp32/machine_pin.c#L106

With this `gpio_install_isr_service(0);` all pin interrupts share the same priority no distinction is possible.

[ESP-IDF doc](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/gpio.html#_CPPv424gpio_install_isr_servicei)

Not sure which priority will be used with `0` though. Probably this could be raised to [`ESP_INTR_FLAG_LEVEL3`](https://github.com/espressif/esp-idf/blob/master/components/esp32/include/esp_intr_alloc.h#L43)

projectgus, 2025-01-07:
Some thoughts on this, after looking at the newer linked issue:

* Since 2018, `hard` parameter is now documented for `machine.irq()` (although support is still patchy across ports).
* I don't think there is any blocker to this feature being added to the esp32 port if someone submits a PR with it (ideally with an automated test included, but having a PR tested by author would still be enough I think.)