micropython/micropython #4221: Theoretical remark regarding mutex in the STM32 port

slavaza, 2018-10-08:
Function pyb_mutex_lock(.) may be unreliable. The execution context switching is performed by setting the interrupt request flag PendSV. In the manuals of the processor do not defined the processing time for this request. In asynchronous machines, with a cache, there is no guarantee that the interruption will occur on the next instruction, after executing the instruction for setting the request flag, even with the highest priority. In addition, this may depend on the implementation of the version of the hardware. Perhaps it would be better to use the SVC instruction for confident context switching. Pay attention to this, please.

dpgeorge, 2018-10-08:
> The execution context switching is performed by setting the interrupt request flag PendSV. In the manuals of the processor do not defined the processing time for this request.

The idea with the code is that setting PendSV will not trigger the interrupt because interrupts are disabled.  It just sets a flag that the interrupt is pending, and the setting of this flag happens immediately.  When interrupts are re-enabled the Cortex-M will check if any interrupts are pending and schedule them if they are; in this case the PendSV is pending and it should schedule that straightaway.  At least, that is how the code was designed to work.  It's possible there could be a bug, if the Cortex-M does not schedule the PendSV after the IRQ priority is raised back to the highest level (lowest priority).

slavaza, 2018-10-08:
Yes I understand this. PendSV is a good idea. But I did not say that this is a bug. I just noticed on exist of the uncertainty of the delay in executing an interrupt after setting the flag. For mutex function it is critical, because can be executed the instructions from a critical section of the mutex, before PendSV interruption occurs.

jonnor, 2024-07-27:
Is this something that should be fixed, or can this issue be closed?

projectgus, 2024-11-20:
I think this is worth looking into, although it's probably not an issue currently. The ARMv7-M Architecture Reference Manual says under "A2.4.1 System Related Events":

> A service (system) call is used by an application which requires a service from an underlying operating system.
The service call associated with PendSV executes when the interrupt is taken. For a service call which executes
synchronously with respect to program execution use the SVC instruction.

I couldn't find a explanation of asynchronous exceptions in the architecture manual, but the Cortex-M4 and Cortex-M7 Devices Generic Users Guide manuals section 2.3.2 Exception Types both repeat the classification of PendSV as asynchronous and say:

> For an asynchronous exception, other than reset, the processor can execute another instruction
between when the exception is triggered and when the processor enters the exception handler

The very next thing that pyb_mutex_lock does is set the IRQ priority to enable the context switch to another task while this one blocks:

```
        // thread switch will occur after we enable irqs
        SCB->ICSR = SCB_ICSR_PENDSVSET_Msk;
        RESTORE_IRQ_PRI(irq_state);
        // when we come back we have the mutex
    }
    return 1; // have mutex
```

If this ends up as a single instruction to store to BASEPRI (or similar) then I think the CPU might execute one more instruction after that before the PendSV interrupt actually triggers (unsure about this detail, but presumably PendSV may not be active at the time the BASEPRI store completes).

In non-LTO builds there is always a function return after this, so worst case the PendSV triggers during this which seems (probably) harmless. But in LTO builds we can't guarantee the compiler won't inline something else here (it probably won't as this function is called from lots of places, but it might.)

So I think a sufficient, if slightly hacky, insurance step would be to place a "nop" after writing ICSR here, just in case.