micropython/micropython #4224: Zephyr Build nrf52840_pca10056

Craigzyc, 2018-10-08:
I posted this on the forum but it does seem to be a legitimate issue based on the error message.

I try to build the zephyr port for nrf52840_pca10056 using this command: 
`make BOARD=nrf52840_pca10056
 `

and it seems to build pretty much to the end but fails. The following error message is shown:
`[ 96%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      214968 B         1 MB     20.50%
            SRAM:       48628 B       256 KB     18.55%
        IDT_LIST:         136 B         2 KB      6.64%
/home/parallels/micropython/ports/zephyr/libmicropython.a(modzephyr.o): In function `mod_stacks_analyze':
/home/parallels/micropython/ports/zephyr/modzephyr.c:46: undefined reference to `k_call_stacks_analyze'
collect2: error: ld returned 1 exit status
zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:104: recipe for target 'zephyr/zephyr_prebuilt.elf' failed
make[3]: *** [zephyr/zephyr_prebuilt.elf] Error 1
make[3]: Leaving directory '/home/parallels/micropython/ports/zephyr/outdir/nrf52840_pca10056'
CMakeFiles/Makefile2:642: recipe for target 'zephyr/CMakeFiles/zephyr_prebuilt.dir/all' failed
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[2]: Leaving directory '/home/parallels/micropython/ports/zephyr/outdir/nrf52840_pca10056'
Makefile:83: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/home/parallels/micropython/ports/zephyr/outdir/nrf52840_pca10056'
Makefile:77: recipe for target 'all' failed
make: *** [all] Error 2`

I can build all examples in zephyr without issue and I can build micropython for zephyr using ./make-minimal but the full build fails.

Craigzyc, 2018-10-08:
If I comment out the line causing the issue it builds and flashes fine. What are the repercussions?

Starting at line 45 in modzephyr.c:
`STATIC mp_obj_t mod_stacks_analyze(void) {
    //k_call_stacks_analyze();
    return mp_const_none;
}`

pfalcon, 2018-10-08:
@Craigzyc, Zephyr port development/maintenance happens in https://github.com/pfalcon/micropython , then synced back to upstream as bandwidth allows (mostly bottlenecked by slow processing of PRs on upstream's side).

That particular issue is being addressed there (WIP).

closed 2018-10-08 by Craigzyc