micropython/micropython #4225: esp8266 uart.read() problem

cemox, 2018-10-08:
Hi,  I am trying to read from uart(0) on nodemcu lolin board. In the below code, I configured uart(1) for sending data and uart(0) for receiving. Both uarts are initiated successfully at 115200 baut. 

I checked uart1 tx pin (i.e. d4-gpio2 on the board) to a serial module,  checked it with cutecom, it works ok. It sends the text at intervals as intended. I ran the code directly from memory (i.e. main.py) and powered the board externally, no usb connected to board. Could not read from the uart0 rx d0 pin (gpio3).
 
`import time
import uselect
from machine import UART

uart0 = UART(0)
uart1 = UART(1)
print(uart0)
print(uart1)

print("starting")
uart1.write("send me something")
uart0.write("waiting for data")

while (True):
    # if (uart0.any() > 3):
    uart1.write("test data")
    time.sleep(0.5)
    # print("reading")
    a = uart0.read(32)
    print(a)
    if a is not None:
        print(a)
        uart0.write('received data')
`

robert-hh, 2018-10-08:
By default, uart 0 is assigned to the REPL prompt. Actually this happens in boot.py. You can read and write with sys.stdin.read() and sys.stdout.write. If you want to use it as a UART, you have to deassign it from REPL, with uos.dupterm(None), or remove the dupterm statement from boot.py.
This is the intended behaviour, not a bug. So this question may better be asked at the forum.micropython.org.

robert-hh, 2018-10-09:
I made a short test code for uart0 read, and it works. The major difference is the second parameter on dupterm.
```
from machine import UART
import uos

uos.dupterm(None, 1)
uart = UART(0, 115200)
uart.write(b"The quick brown fox jumps over the lazy dog\r\n")
ch = b""
while ch != b"q":
    if uart.any():
        ch = uart.read()
        uart.write(ch)

uos.dupterm(UART(0, 115200), 1)
```

dpgeorge, 2018-12-05:
I assume the above explanation solved this issue.

closed 2018-12-05 by dpgeorge

mikronauts, 2019-09-16:
I tried this with the latest official ESP32 micropython, and it did not work - I got a message that uart 0 was reserved for repl.

There should be a way to release uart 0 ... sometimes I need to access uart 0.

robert-hh, 2019-09-16:
On ESP32 you can use UART1.

mikronauts, 2019-09-16:
Sometimes I need three serial ports.

I found after posting here

https://forum.micropython.org/viewtopic.php?f=18&t=6954

I will try it today, if it works problem solved.

robert-hh, 2019-09-16:
You can read & write from uart0 with sys.stdin.read() and sys.stdout.write(). That will however perform a UTF-8 coding. For raw data you have to use sys.stdin.buffer.read() and sys.stdout.buffer.write(). To disable checks for Ctrl-C, call micropyton.kbd_intr(-1) at the beginning.

mikronauts, 2019-09-16:
Thank you. Previously sys.stdout.write worked for me, but sys.stdout.read did not. 

Appreciate the hint on how to disable ctr-c!

dev-lexnotionus, 2019-10-26:
> I made a short test code for uart0 read, and it works. The major difference is the second parameter on dupterm.
>
> ```
> from machine import UART
> import uos
>
> uos.dupterm(None, 1)
> uart = UART(0, 115200) [...]

ESP8266 LOL1n v0.1 board, MicroPython on RPi 3B+, uPyCraft v1.1

could ```uos.dupterm()``` please be explained:
```uos.dupterm(None, 1)``` does this drop out of python, into serial?
```uos.dupterm(UART(0, 115200), 1)``` and this climbs back to python?

I have read the manual on ```uos.dupterm```, but some terms are unknown to me... but I have run the tests of changing things, until it breaks, and then fixing it again... and the test point to my above layman statements.