micropython/micropython #4238: [STM32] DTCM buffer incorrect sized for processor type

rolandvs, 2018-10-14:
In `flashbdev.c`[location](https://github.com/micropython/micropython/blob/de71035e02e11bf89b09d2ec6292518d29f0f0e7/ports/stm32/flashbdev.c#L79) and below the used DTCM is 64kByte, however the STM32F76xxx and STM32F77xxx feature a DTCM of 128kByte, whereas the STM32F75xx and STM32F74xx have a 64kByte DTCM.

dpgeorge, 2018-10-15:
It shouldn't really matter the total size of the DTCM.  The flashbdev.c code only uses 32k of it for these F7xx devices.

What is important is that the relevant RAM areas are reserved in the corresponding .ld files.

rolandvs, 2018-10-15:
I need to check again, but in order to use 128kByte sectors for internal Flash drive I should have the ability to modify the whole block and need the 128kByte DTCM...