micropython/micropython #4239: (Re)application for commit access

pfalcon, 2018-10-15:
There appears to be rather slow pull request processing by the maintainer, for example, only of my recent PRs, there're 13 waiting in queue (and I could easily add a dozen more), including such trivial as https://github.com/micropython/micropython/pull/4216 or https://github.com/micropython/micropython/pull/4217 . This is of course no surprise and all to be expected, especially with the sole maintainer, based on the experience with other open source projects, and something I explicitly was set to address (i.e. provide timely response) when I was a comaintainer.

Given that there's still a common base between mainline and many forks (including mine), and the fact that forks (including mine) to concentrate on their dedicated areas, while the mainline to get more active maintenance and bugfixes in already established/common base.

Based on that, I would to apply for the commit access to be able to fix bugs/maintain/optimize code in the following areas, which originally were designed/implemented/maintained by me:

* Various extmod/ modules (uctypes is one example needing continuing development/"finishing")
* stream subsytem
* potentially, parts of instance types' type subsystem
* Zephyr port
* unix port
* continuing reduction of the minimal binary size
* continuing improvement to the test infrastructure
* upip package manager
* library modules and submodules: uzlib, re1.5, axtls
* documentation
* etc.

pfalcon, 2018-10-22:
Ping.

pfalcon, 2018-11-02:
@dpgeorge, ping ;-).

dpgeorge, 2018-11-28:
> I would to apply for the commit access to be able to fix bugs/maintain/optimize code in the following areas

I'm sorry, I cannot agree to this request.  There have been many discussions here that show our goals/vision/philosophy for this project are too different and our opinions clash too strongly.  So it would be counter productive to grant commit access, continually disagreeing on each other's commits.

pfalcon, 2018-11-28:
> There have been many discussions here that show our goals/vision/philosophy for this project are too different and our opinions clash too strongly. 

Well, that's weird way to put it, and a pure wonder how we worked together for 4 years before then. How I see it, is that together we cover with our visions much wider area than someone could do alone.

> So it would be counter productive to grant commit access, continually disagreeing on each other's commits.

Well, I have my fork, and going to have it, for all things that you disagree with (or I do, but surprisingly, there were too few on them during the last year, hence this request in the first place). Beyond that, there's a lot of common infrastructure, which needs active maintenance/development, which you alone just can't keep up with.

closed 2018-12-04 by dpgeorge