micropython/micropython #4245: unable to match

pacmac, 2018-10-17:
Also posted this on the forum but no replies, so I thought I would ask here.

https://regex101.com/r/vouYnV/1

Any idea why my regex won't find & remove this hex value embedded in carriarge return / line feeds :

```
>>> re.sub(r'^\\r\\n[0-9,a-f]+\\r\\n','','\r\nfe5\r\nabcdefghxxxxx')
'\r\nfe5\r\nabcdefghxxxxx'
```

on regex101 it matches but I can't get it to work in uPython !

what it should return is:
`abcdefghxxxxx`

This is used to removed 'chunked' data from http 1.1 requests.

dmazzella, 2018-10-17:
try this:

`re.sub('^\r\n[0-9,a-f]+\r\n','','\r\nfe5\r\nabcdefghxxxxx')`

pacmac, 2018-10-17:
Just tried it in Python 2.7 and Python 3 and it works as expected in both !

```
>>> re.sub('^\\r\\n[0-9,a-f]+\\r\\n','','\r\nfe5\r\nabcdefghxxxxx')
'abcdefghxxxxx'
>>> 
```

pacmac, 2018-10-17:
@dmazzella - Thanks, yes that works without using r strings:

```
>>> re.sub('^\r\n[0-9,a-f]+\r\n','','\r\nfe5\r\nabcdefghxxxxx')
'abcdefghxxxxx'
>>> 
```

dpgeorge, 2018-10-17:
You have to be careful with escape codes and distinguish that from a literal backslash.  Eg:
```python
>>> b'\r'.hex() # a single char
'0d'
>>> b'\\r'.hex() # literal slash and r
'5c72'
>>> rb'\r'.hex() # r-strings don't escape slash
'5c72'
>>> rb'\\r'.hex()
'5c5c72'
```
In that regex101 link the \r and \n are not control characters but rather literal slashes and r/n.  If the regex101 is fixed to use control characters it should match CPython.

closed 2018-10-17 by dpgeorge