micropython/micropython #4250: TypeError:argument has wrong type

bricedenicedu69, 2018-10-18:
Hi, I am new with Python and currently trying to display values on a LED screen (it is a RiT but I intend to use the SSD1306 one which seems appropriate) using a I2C connection. Thus, I first do a : from machine import I2C. I also connected the SDA, SCL, GND and VCC of the screen to the corresponding ones on my Pycom device. However, when I try to init the I2C bus it does not work because apparently, the arguments of the parameters do not have the right type.

scl=Pin('P10',Pin.IN)
sda=Pin('P9',Pin.IN)
i2c=I2C.init(sda,scl)

This is the block of code which does not work. My question is : why do I get this error message when it is written on the I2C class documentation (which you can find here : https://docs.micropython.org/en/latest/library/machine.I2C.html) that sda and scl should be pin objects, which I did. Furthermore, if I just write : i2c=I2C(sda,scl) I got the following message : TypeError: can't convert Pin to it.
Thank you all for your help

robert-hh, 2018-10-18:
You are using the syntax of the Pycom variant of Micropython. For defining a Pin object, you should look here: https://docs.micropython.org/en/latest/library/machine.Pin.html?highlight=machine%20pin, unless you use a Pycom module.
Besides that, questions like these are to be raised at forum.micropython.org for the micropython.org variant or forum.pycom.io for Pycom products and dialect.
This place here is dedicated to bugs.

bricedenicedu69, 2018-10-19:
Thanks for answering and sorry for posting this message in the wrong place. Indeed, I use a Pycom device (Sipy more precisely).

closed 2018-10-19 by dpgeorge