micropython/micropython #4256: auto-completion unexpectedly active in input()? (bug?)

frankseide, 2018-10-20:
Hi, I just noticed that the `input()` function seems to have Python code completion enabled. I suspect that this is not intentional?

(This is running from the Windows build, I have not tried it on the actual microcontroller.)

```
>>> a=input()
__
__class__       __name__
__class__
>>> a
'__class__'
```

dpgeorge, 2018-10-23:
> Hi, I just noticed that the `input()` function seems to have Python code completion enabled. I suspect that this is not intentional?

It is intentional because that is how normal Python works: the `input()` function has full readline support, including history and tab completion.

frankseide, 2018-10-23:
Interesting. Yes, it does that indeed; never noticed. Having history and tab completion in `input()` is great, but input-completion from user programs w.r.t. Python syntax feels like that's a bug in normal Python that you are replicating.

Anyway, thanks for your great work with MicroPython!

closed 2018-10-23 by frankseide