micropython/micropython #4262: HW SPI type does not match machine.SPI

MrSurly, 2018-10-23:
```python
import machine
s = machine.SPI(1, 10000000, sck=machine.Pin(18), mosi=machine.Pin(19))
type(s) = machine.SPI
False

## Huh?

type(s)
<class 'SPI'>
type(machine.SPI)
<class 'type'>
machine.SPI
<class 'SoftSPI'>
```

Not sure how to resolve this in C code ...

Present workaround is 
```python
isSpi = 'SPI' in repr(type(s))
```

hardkrash, 2018-12-07:
What are you trying to do?
Are you trying to detect if s is an instance of machine.SPI?
You'll want the `isinstance(obj, class)` or `issubclass(class, parent_class)` methods.

But more often the pythonic answer is to use try and except and "not look before you leap"

What do you mean with "Not sure how to resolve this in C code ..."?

MrSurly, 2018-12-17:
> What are you trying to do?

Have consistency in the interface.  The `type` returned for other HW objects is correct.

```python
import machine
>>> u = machine.UART(1, 9600)
I (5396) uart: ALREADY NULL
>>> type(u) == machine.UART
True
>>> i = machine.SPI(1, 10000000, sck=machine.Pin(18), mosi=machine.Pin(19))
>>> type(i) == machine.SPI
False
>>>
```

> But more often the pythonic answer is to use try and except and "not look before you leap"

This is validation code for a class `init` having it fail later is not a use case.

> What do you mean with "Not sure how to resolve this in C code ..."?

I mean how to fix it in the underlying C code.  I've looked a bit at the type system for MP, was hoping @dpgeorge could give me a hint in the right direction.

closed 2020-10-01 by dpgeorge