micropython/micropython #4267: unix: -X emit= command line option is not global

pfalcon, 2018-10-24:
One would expect  that `-X emit=` option "sets the default code emitter" as said in help. But it's not, it affects only the script passed on the command line. It doesn't affect imports, and by extension doesn't affect -m (execute from module) command line option.

This is confusing and less useful than if it was truly global.

This was found running `micropython -X emit=native -m pystone` vs `micropython -m pystone` and wondering why emit=native case *seemed* slower.

The story doesn't end here, I'll submit separate ticket.

dpgeorge, 2018-10-30:
Thanks for reporting this, it's something to fix.

A related point which will affect how this issue is fixed is how to set the default emitter on (bare metal) targets that don't have the equivalent of "-X emit=".  There is already `micropython.opt_level([value])` for setting the optimisation level (similar to command line -O<value>), so it probably makes sense to either add another argument to this function to set the default emitter, or add another function like `micropython.emitter([value])`.

pfalcon, 2018-10-31:
Well, I guess this setting definitely should start as global. Someone wanting to have it e.g. per-thread would need to present compelling usecase. That alone would allow to fix it for unix port.

> micropython.opt_level([value])

Yeah, I definitely think it makes sense to extend this function, to save on boilerplate data structures/code. Of course, after renaming it to e.g. ``micropython.comp_options()``, or if we're brave enough, to just ``micropython.options()``. Just take another param, or if we want to support querying values, and save it once at for all, then following network.config param pattern. (I'd personally think that just 2nd arg and not bothering with querying is enough for now.)

dpgeorge, 2019-08-28:
The "-X emit=..." command line option to unix, windows and mpy-cross is now global, see af20c2ead3e9bb397fdf89e316aa78b56f165013

closed 2019-08-28 by dpgeorge