micropython/micropython #4268: unix: Native emitter is slower than bytecode on "PC" hardware

pfalcon, 2018-10-24:
Once figuring #4267 situation, I proceeded with actual native vs bytecode testing, and here's the picture:

~~~
$ micropython -X emit=native pystone.py 
Pystone(1.2) time for 50000 passes = 0.675
This machine benchmarks at 74074.1 pystones/second
$ micropython  pystone.py 
Pystone(1.2) time for 50000 passes = 0.614
This machine benchmarks at 81433.2 pystones/second
~~~

Of course, that's a typical irony of native codegeneration, but we should understand why. And let me guess, that because of the lack of caching of dictionary lookups. So, ways to address that should be pondered.

dpgeorge, 2018-10-26:
> And let me guess, that because of the lack of caching of dictionary lookups.

Yes that is true.  See some previous discussion here: https://github.com/micropython/micropython/issues/1041#issuecomment-90742337

pfalcon, 2018-10-26:
> See some previous discussion here: #1041 (comment)

D'oh, so I knew that it runs, and then forgot, and trying to do it doesn't work as expected (#4267). There's a lot to do to make this suitable for day to day use instead (and not doing that means semi-wasting years of effort).

Feel free to close and let me know, if this ticket is not useful given the discussion in #1041.