micropython/micropython #4269: ESP32: Hangs when reopening WiFi connection

nickzoic, 2018-10-25:
This is a copy of micropython/micropython-esp32#167 reported by @peterhinch to bring it across to the main repo.

Summary: when waiting on network events, such as in a loop like:
```
while not s.isconnected(): pass
```

As far as I understand it, the network background tasks never get run, so the loop never ends. This isn't as intended.

As a workaround, if `utime.sleep_ms(11)` or `machine.idle()` is called, this yields the micropython task and allows other tasks to run.  This shouldn't be left to the Python programmer though.

dpgeorge, 2018-12-27:
Thanks for copying this across from the old repo, and thanks @nickzoic for the following hint how to fix it:

> We also haven't really answered the question of why it works on the first reset ... what changes?

So it works fine after a hard reset, but not after a soft reset.  That really points to it being a problem with soft reset, in combination somehow with FreeRTOS scheduling.  Since including a long sleep in the waiting works around the issue, this further points to an problem with soft reset and how idling works.

After some investigation I disabled threading and it fixed the problem.  Then I just disabled the GIL and that also fixed the problem.  Thus it's a problem with soft reset and the GIL, in combination with FreeRTOS scheduling.

In the end the fix is simply to release the GIL when deinitialising the uPy runtime, which is now done in fa50047bbc25a2db53ef2c38b7735a4dedb23b21 .  This fix makes sense regardless of the issue here.

In summary: I think what was happening is that the main uPy task still had hold of the FreeRTOS mutex (corresponding to the GIL) upon soft reset and this put the uPy task on a special FreeRTOS scheduling queue (perhaps raising its priority) which prevented network events getting delivered to it after soft reset.  There GIL mutex was reinitialised upon soft reset but this didn't help because FreeRTOS had no way to know that the old one was no longer used.

closed 2018-12-27 by dpgeorge

mattytrentini, 2018-12-27:
That's great news, thanks Damien! I'm sure @peterhinch will be happy too; this issue had been bothering him for a long time. :)