micropython/micropython #4270: SD card Nucleo_F746ZG

ianaherne, 2018-10-25:
Hi 
I have been testing Micropython with some STM32 Nucleo Boards and adding SD card functionality by wiring an SD card directly to the board and modifying  mpconfigboard.h accordingly as per below link.

https://www.kaltpost.de/?p=2199

I have done this successfully with F401RE and L476RG and can run python scripts from the SD card.

I have tried this also with a Nucleo F746ZG modifying mpconfigboard.h  in the same way but with no success.

on start up I get the warning PYB: cant mount SD card.

I noticed in the pins.csv file for the F746ZG there were no SDMMC pins listed so I had to add them manually in order to compile which would suggest that SD card functionality has never been tested for this particular processor.

logic analyzer shows clock signal and cmd data but no data on any of the data pins.

The Not working F746ZG clock signal is 400KHz for a short period and then stops.

The working F401RE shows clock signal of 400 khz then 8MHZ when it starts to transfer data from SD Card.

has anyone else attempted this with this particular board/Processor?

dpgeorge, 2018-10-30:
The SD card has been well tested with F722, F733 and F767 MCUs.  It should work on either SDMMC1 or SDMMC2.  It looks like the F746 only has SDMMC1, and it should be enough to define the following in mpconfigboard.h:
```c
#define MICROPY_HW_SDMMC2_CK                (pin_C12)
#define MICROPY_HW_SDMMC2_CMD               (pin_D2)
#define MICROPY_HW_SDMMC2_D0                (pin_C8)
#define MICROPY_HW_SDMMC2_D1                (pin_C9)
#define MICROPY_HW_SDMMC2_D2                (pin_C10)
#define MICROPY_HW_SDMMC2_D3                (pin_C11)
#define MICROPY_HW_SDCARD_DETECT_PIN        (pin_A0) // change this to the correct pin
#define MICROPY_HW_SDCARD_DETECT_PULL       (GPIO_PULLUP)
#define MICROPY_HW_SDCARD_DETECT_PRESENT    (GPIO_PIN_RESET)
```

dpgeorge, 2018-12-04:
I assume the above suggestion solved the problem.  If not feel free to reopen.

closed 2018-12-04 by dpgeorge