micropython/micropython #4272: Micropython allows access to 'nonlocal' variables without 'nonlocal' keyword

ddiminnie, 2018-10-25:
It's not clear to me whether the behavior described below is a bug or a design decision (possibly for efficiency).  It is a divergence from CPython behavior.  (My apologies if this has been discussed elsewhere - a few quick searches did not turn up anything for me.)

Consider the following (sloppy) code:
```python
def outer():
    x = 1
    def inner():
        x += 1
        print(x)
    inner()
```
On CPython calling this function triggers an 'UnboundLocalError':
```
>>> outer()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<stdin>", line 6, in outer
  File "<stdin>", line 4, in inner
UnboundLocalError: local variable 'x' referenced before assignment
```
In MicroPython/CircuitPython calling this function prints a value:
```
>>> outer()
2
```
Micropython does not have an UnboundLocalError exception... so perhaps a NameError could be raised instead?...

... or the 'MicroPython differences from CPython' documentation could be updated to add this difference (if it is, in fact, a design decision for efficiency).

dpgeorge, 2018-10-26:
Thanks for the report.  Indeed it is a bug.

Consider the following:
```python
def f():
    x = 1
    def g():
        print(x) # at this point is x a local or implicit nonlocal?
        x = 1 # now we know that x is a local
```
At the first use of `x` in `g` it is unknown whether `x` is a closed over variable (nonlocal) or a local.  MicroPython is currently a bit too eager here and because it finds `x` in the outer scope of `f` it assumes it is closed over.  But it's not until later on in the function `g` that the compiler sees that `x` is assigned to and is therefore a local variable, and by this point the decision was already made that it was (implicitly) nonlocal.  (If the assignment to `x` in `g` is not there then `x` should be an implicit nonlocal.)

So, the compiler needs to delay its decision as to whether a variable is made implicitly nonlocal or kept local.

dpgeorge, 2018-10-27:
Should be fixed by 9201f46cc8e92231f0f6c92e4d56befbc150f72c

closed 2018-10-27 by dpgeorge