micropython/micropython #4276: NRF - Ctrl-E Fatal Exception

diginfo, 2018-10-28:
Pasting any simple multi line code appears to crash the device:

This is just a simple 3 line to paste for testing:

```
print(1)
print(2)
print(3)
```

```
MicroPython v1.9.4-674-g27ca9ab-dirty on 2018-10-28; holyiot_18010 with NRF52840
Type "help()" for more information.
>>> 
paste mode; Ctrl-C to cancel, Ctrl-D to finish
=== def bytes_to_str(bytes):
=== FATAL: uncaught exception 20003dd4
MemoryError: 
```

If I pasted these directly into the REPL without CTRL-E then there's no problem.

glennrub, 2018-11-03:
Hi @diginfo,

I believe i have seen this before, but have forgotten what triggers this. I wonder if @dpgeorge is onto something. To isolate this a bit, does this error occur bluetooth UART REPL or regular UART REPL? and if on regular UART, is flow control  on UART enabled or disabled?

diginfo, 2018-11-03:
Thanks for helping

It is when using the uart with no flow control 

I am using holyiot 52840

jonnor, 2024-07-27:
Does this still happen in the lastest versions of MicroPython, such as 1.23+?

robert-hh, 2024-07-27:
Ctrl-E for paste mode works meanwhile. Tested on a NRF52840 board.

closed 2024-07-27 by robert-hh