micropython/micropython #4280: esp32 port installation attempt on Kubuntu fails

pj4grr, 2018-10-31:
- esp-idf configures, compiles, flashes and monitors the examples on a esp32 devkit v4 connected to kubuntu via /dev/ttyUSB0 (blink works, wifi station responds to ping).
- unix port of micropython makes normally and works.

Attempting to make esp32 port, I face the following problem, although both the esp-idf and micropython structures are cloned recently:

$ make
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/jkido/esp/esp-idf
Current git hash:   7efbe8178b9e25ac404c02cf6908474ab3918995
Supported git hash: 30545f4cccec7460634b656d278782dd7151098e
LINK build/bootloader.elf
build/bootloader/libbootloader_support.a(bootloader_init.o):(.literal.bootloader_init+0xe4): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_init.o): In function `bootloader_init':
bootloader_init.c:(.text.bootloader_init+0x407): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_clock.o):(.literal.bootloader_clock_configure+0x14): undefined reference to `rtc_clk_init'
build/bootloader/libbootloader_support.a(bootloader_clock.o): In function `bootloader_clock_configure':
bootloader_clock.c:(.text.bootloader_clock_configure+0x76): undefined reference to `rtc_clk_init'
collect2: error: ld returned 1 exit status
Makefile:830: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1

pj4grr, 2018-10-31:
using make V=1, I obtain:

$ make V=1
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/jkido/esp/esp-idf
Current git hash:   7efbe8178b9e25ac404c02cf6908474ab3918995
Supported git hash: 30545f4cccec7460634b656d278782dd7151098e
LINK build/bootloader.elf
xtensa-esp32-elf-gcc  -nostdlib -L/home/jkido/esp/esp-idf/lib -L/home/jkido/esp/esp-idf/ld -u call_user_start_cpu0 -Wl,--gc-sections -static -Wl,-EL -Wl,-Map=build/bootloader.map -Wl,--cref -T /home/jkido/esp/esp-idf/components/bootloader/subproject/main/esp32.bootloader.ld -T /home/jkido/esp/esp-idf/components/bootloader/subproject/main/esp32.bootloader.rom.ld -T /home/jkido/esp/esp-idf/components/esp32/ld/esp32.rom.ld -T /home/jkido/esp/esp-idf/components/esp32/ld/esp32.rom.spiram_incompatible_fns.ld -T /home/jkido/esp/esp-idf/components/esp32/ld/esp32.peripherals.ld -o build/bootloader.elf  -Wl,--start-group build/bootloader//home/jkido/esp/esp-idf/components/soc/esp32/rtc_clk.o build/bootloader//home/jkido/esp/esp-idf/components/soc/esp32/rtc_time.o build/bootloader//home/jkido/esp/esp-idf/components/soc/esp32/cpu_util.o build/bootloader//home/jkido/esp/esp-idf/components/bootloader/subproject/main/bootloader_start.o -Lbuild/bootloader -lbootloader_support -llog -lspi_flash -lmicro-ecc -L/home/jkido/esp/esp-idf/components/esp32/lib -lrtc -L/home/jkido/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/ -lgcc -Wl,--end-group
build/bootloader/libbootloader_support.a(bootloader_init.o):(.literal.bootloader_init+0xe4): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_init.o): In function `bootloader_init':
bootloader_init.c:(.text.bootloader_init+0x407): undefined reference to `rtc_wdt_disable'
build/bootloader/libbootloader_support.a(bootloader_clock.o):(.literal.bootloader_clock_configure+0x14): undefined reference to `rtc_clk_init'
build/bootloader/libbootloader_support.a(bootloader_clock.o): In function `bootloader_clock_configure':
bootloader_clock.c:(.text.bootloader_clock_configure+0x76): undefined reference to `rtc_clk_init'
collect2: error: ld returned 1 exit status
Makefile:830: recipe for target 'build/bootloader.elf' failed
make: *** [build/bootloader.elf] Error 1

dhylands, 2018-10-31:
These two lines show the problem:
```
Current git hash: 7efbe8178b9e25ac404c02cf6908474ab3918995
Supported git hash: 30545f4cccec7460634b656d278782dd7151098e
```
You need to use the 30545f4cccec7460634b656d278782dd7151098e version of the ESP IDF

pj4grr, 2018-11-01:
Thank you very much, but since I am a newbie, could you please help me a bit with github?
I am trying:
git clone https://github.com/espressif/esp-idf/tree/30545f4cccec7460634b656d278782dd7151098e
but cloning reports "not found"
Thank you in advance.

closed 2018-11-01 by pj4grr