micropython/micropython #4282: PR CI testing regressed

pfalcon, 2018-11-01:
As was mentioned in https://github.com/micropython/micropython/pull/4261#issuecomment-432368939 , something went wrong, and there's no longer an integration between Travis CI runs and pull requests. That makes assessing quality of PRs quite hard.

pfalcon, 2018-11-01:
This appears to be last PR where integration yet worked: https://github.com/micropython/micropython/pull/4169

dpgeorge, 2018-11-02:
Apparently GitHub services were deprecated on 1 Oct 2018, so that's why Travis for PRs stopped working then.  They are replaced with GitHub WebHooks, and/or GitHub Apps.  Travis CI .org (what is used by this project) still only supports services (as far as I can tell).  Travis CI .com supports newer WebHooks and Apps.  The options would be:
1. wait until Travis CI .org supports WebHooks or Apps
2. change to use Travis CI .com, which might be possible because they seem to allow free accounts for open source projects (https://blog.travis-ci.com/2018-05-02-open-source-projects-on-travis-ci-com-with-github-apps) but they don't yet have the possibility to migrate from .org to .com

dhalbert, 2018-11-02:
https://github.com/adafruit/circuitpython is still using GitHub services for travis-ci.org integration, and it's still working for us. So I'm not sure what's going wrong for you; you might want to check on the travis side as well. Or perhaps one of your tokens expired?? Here are some screenshots:

[image]

### **In some cases we have a user specifed:**
[image]

### **But for most of our libraries, the settings look like this:**

[image]

dhalbert, 2018-11-02:
I inquired about travis-ci migrating from .org to .com in August, and got this response (edited to the most interesting part):

>So, repositories cannot be migrated automatically- they require us to flip a switch on the backend to actually complete the migration, so you don't need to worry about anything there. Currently, the manual migration does not preserve build history or settings, so it's opt-in only. Repositories will be migrated automatically some time before the end of the year with build history and settings intact- so if losing those is not acceptable, you can just hang tight until we begin the automatic migration.

dpgeorge, 2018-11-02:
Thanks @dhalbert for the info.  The Travis settings for this repo do look similar to those you posted.  I tried to refresh the token but GitHub gave me an error, so I'm not sure what to do.

dhalbert, 2018-11-02:
Our settings in travis. This would be in https://travis-ci.org/micropython/micropython/settings. Note the "Build pushed pull requests" is on. I don't see anything unusual in your `.travis.yml`. I'm seeing recent PR builds of yours here, so I'm not sure what's not working for you now: https://travis-ci.org/micropython/micropython/pull_requests

[image]

dpgeorge, 2018-11-03:
> Note the "Build pushed pull requests" is on.

Yes that's enabled, the settings are the same as yours.

> I'm seeing recent PR builds of yours here, so I'm not sure what's not working for you now

Yes, Travis is actually building all PRs, the issue seems to be that it is not posting the results back to GitHub, so that's why I think it's something to do with GitHub services being deprecated.

dhalbert, 2018-11-03:
Hmm. We are still seeing travis builds in progress when we do a PR, and then we see a check or X as before in the PR conversation. For example: "All checks have passed" in this recent PR: https://github.com/adafruit/circuitpython/pull/1313. I looked and I don't see that in the most recent PR in yours.

dpgeorge, 2018-12-05:
> Apparently GitHub services were deprecated on 1 Oct 2018, so that's why Travis for PRs stopped working then.

It looks like this was not actually the cause.  The problem was that authentication settings changed for the micropython github organisation and that left the CI tools unable to push data to PRs.  It was just a coincidence that this happened around the same time as github deprecating its integration services.

I've now reset the authentication and Travis and Appveyor are now working again.  I'm unsure about Coveralls.

dpgeorge, 2018-12-10:
Coveralls seems to be working again, so everything should be back to normal.

closed 2018-12-10 by dpgeorge