micropython/micropython #4284: help(webrepl) incorrectly executes functions

Hermann-SW, 2018-11-02:
man(), pwd(), clear() and pwd() should as "... -- <function>" instead of being executed:
```
>>> help(upysh)
object <module 'upysh'> is of type module
  man -- 
upysh is intended to be imported using:
from upysh import *

To see this help text again, type "man".

upysh commands:
pwd, cd("new_dir"), ls, ls(...), head(...), cat(...)
newfile(...), mv("old", "new"), rm(...), mkdir(...), rmdir(...),
clear

  ls --     3208 FBConsole.py
     690 boot.py
     342 dup.py
     335 dup2.py
     346 dup3.py
       0 empty
   10414 isalpha.py
    7095 isword.py
    7077 isword_.py
   10781 iswordold.py
    1681 keypad.py
    6146 keypad_timer.py
     149 myio.py
    5542 p0.py
     293 srch.py
    4585 ssd1306.py
       6 tst
     557 tst.py
      33 tst.txt
   10120 upysh_.py
   10575 upyshlambda.py
      14 webrepl_cfg.py

  rm -- <function>
  __name__ -- upysh
  cd -- <function>
  Man -- <class 'Man'>
  clear -- 

  pwd -- /
  mkdir -- <function>
  head -- <function head at 0x3ffe8450>
  CLEAR -- <class 'CLEAR'>
  cat -- <function cat at 0x3ffe8470>
  mv -- <function>
  sys -- <module 'sys'>
  os -- <module 'uos'>
  rmdir -- <function>
  PWD -- <class 'PWD'>
  newfile -- <function newfile at 0x3ffe8480>
  LS -- <class 'LS'>
>>> 
```

peterhinch, 2018-11-04:
Should this issue be raised against [Paul's library](https://github.com/pfalcon/micropython-lib)?

jonnor, 2024-09-29:
upysh is not part of core MicroPython. Latest code is in https://github.com/micropython/micropython-lib/tree/394cbfc98a333dd1d4db35fb69379c72c30337f3/micropython/upysh

closed 2024-09-29 by jonnor