micropython/micropython #4285: sys.stdout.write() unconditionally adds \r for any \n in string

Hermann-SW, 2018-11-02:
One might accept replacing "\n" with "\r\n" in case is not prefixed with "\r", but replacing "...\r\n..." with "...\r\r\n..." definitely is a bug:
```
>>> od('c', lambda: sys.stdout.write('xxx\nabc'))
000000   x   x   x  \r  \n   a   b   c
        78  78  78  0d  0a  61  62  63
000008
>>> od('c', lambda: sys.stdout.write('xxx\r\nabc'))
000000   x   x   x  \r  \r  \n   a   b   c
        78  78  78  0d  0d  0a  61  62  63
000009
>>> od('c', lambda: sys.stdout.write('xxx\rabc'))
000000   x   x   x  \r   a   b   c
        78  78  78  0d  61  62  63
000007
>>> 
```

pmp-p, 2018-11-11:
i don't think it's adding anything,  it is just replacing "\n"  by "\r\n" in the stream as it should. why would you send "\r\n" (which is a sequence not a end of line) manually on a terminal line anyway ?

Hermann-SW, 2018-11-11:
Because of pipelining, at least when I reported this issue. At that time I used sys.stdout.write() and os.dupterm to capture the output of any command written to stdout.

I just checked, and my current implementation does not have that problem of ever growing "\r"s per line anymore, because now os.dupterm is not used for pipeing anymore, see below. Anyway, who gives sys.stdout.write the right to add a "\r" for each round of processing an input file and creating a new output file, that will be used as input for another processing?

```
>>> od("tst.txt","c")
000000   f   i   r   s   t  \n   s   e   c   o   n   d   (   )   .  \n
        66  69  72  73  74  0a  73  65  63  6f  6e  64  28  29  2e  0a
000010   T   h   i   r   D  \n           F   o   u   r   t   h   (   )
        54  68  69  72  44  0a  20  20  46  6f  75  72  74  68  28  29
000020  \n
        0a
000021
>>> pipe("tst.txt") | (head,3) | (od,"c") | done
000000   f   i   r   s   t  \n   s   e   c   o   n   d   (   )   .  \n
        66  69  72  73  74  0a  73  65  63  6f  6e  64  28  29  2e  0a
000010   T   h   i   r   D  \n
        54  68  69  72  44  0a
000016
>>> pipe("tst.txt") | (head,3) | (tail,2) | (od,"c") | done
000000   s   e   c   o   n   d   (   )   .  \n   T   h   i   r   D  \n
        73  65  63  6f  6e  64  28  29  2e  0a  54  68  69  72  44  0a
000010
>>> 
```

dpgeorge, 2018-11-12:
> who gives sys.stdout.write the right to add a "\r" for each round of processing an input

This follows CPython behaviour where `sys.stdout` and `sys.stdin` offer "cooked" processing.  Reading from `stdin` will also convert `\r` to `\n`.

Consider how the built-in `print` works: by default it uses `"\n"` as the end of line string, and writes its output to `sys.stdout`.  So to make `print(...)` have sensible behaviour at a raw terminal `sys.stdout` must convert newlines.

It is quite easy to get around this by using `sys.stdout.buffer` and `sys.stdin.buffer` instead.  They don't do any processing.  (On CPython `sys.stdout` is a `TextIOWrapper` while `sys.stdout.buffer` is a `BufferedWriter`.)

Hermann-SW, 2018-11-12:
Thanks for the detailed explanation, especially on "cooked" and the easy workaround.
I will use sys.stdout.buffer whenever I run into this issue again, currently I don't:
[https://github.com/Hermann-SW/micropython-lib#fork-mission-statement](https://github.com/Hermann-SW/micropython-lib#fork-mission-statement)
Closing issue now.
```
>>> pipe(lambda: sys.stdout.write('foo\nbar')) | (od,'c') | done
000000   f   o   o  \r  \n   b   a   r
        66  6f  6f  0d  0a  62  61  72
000008
>>> pipe(lambda: sys.stdout.buffer.write('foo\nbar')) | (od,'c') | done
000000   f   o   o  \n   b   a   r
        66  6f  6f  0a  62  61  72
000007
>>> 
```

closed 2018-11-12 by Hermann-SW