micropython/micropython #4286: can not use onboard LED esp32

derdigge, 2018-11-02:
Hello.

I got ESP32 from azdelivery. It has one onboard led(D1). After installing micropython, its used as uard led?
If i try to set the D1 like usual - machine.Pin(1, machine.Pin.OUT) - the led lits up and device freezes.

Is ther something i can do?

Thanks.

robert-hh, 2018-11-02:
This is a typical forum question. besides that: No, there seems little you can do. If both the LED and the UART/USB bridge are hardwired to GPIO1, then you cannot use them independently.

derdigge, 2018-11-02:
Thanks for your answer, sorry for spaming. forgot to mention, that in arduino ide(C) it just worked for me. Serial was off. I swap to another board. Thanks.

closed 2018-11-02 by derdigge