micropython/micropython #4287: PYBOARD V1.1 LED Green and Yellow blink with no program running after powering on.

jsbyysheng, 2018-11-02:
I design a board with the same hardware configuration as PYBOARD V1.1 but I meet a strange problem. After my board powering on, the LED Green and Yellow blink no matter I put any python code in the flash. Meanwhile, the program in the flash won't execute. After 60s, the board can connect to the computer and I notice the boot.py doesn't execute either( for I put 'import pyb' in it but I cannot use pyb in REPL). When I connect to the computer, I can use REPL without any error and all the functions are normal. I have no idea with this for a long time and I wonder if someone can help me. I will appreciate anyone who offers me help.

dhylands, 2018-11-02:
I normally do the following when trying to debug issues like this:
1 - Set MICROPY_HW_UART_REPL and MICROPY_HW_UART_REPL_BAUD for one of the UARTs 
2 - Connect a serial terminal up to the configured UART
3 - boot up your board. You should get a REPL, and possibly other messages (like hard faults etc) that won't appear on the USB port.

jsbyysheng, 2018-11-03:
Thank you. I will try your advice soon.

jsbyysheng, 2018-11-03:
> I normally do the following when trying to debug issues like this:
> 1 - Set MICROPY_HW_UART_REPL and MICROPY_HW_UART_REPL_BAUD for one of the UARTs
> 2 - Connect a serial terminal up to the configured [...]

I have found the problem. It caused by AD7606 which I connected to the SPI port of the pyboard. The IC
was broken and all is ok after replacing another AD7606. I still appreciate for your advice.

closed 2018-11-03 by jsbyysheng