micropython/micropython #4288: Can not apply advertisment data

linycm, 2018-11-03:
Hi @glennrub ,

  I got a similar issue with https://github.com/micropython/micropython/issues/4179 .

  I test bluetooth example on my NRF board PCA10040, then got error for cannot apply advertisment data. I put some in REPL as below:
>>> from ubluepy import Service, Characteristic, UUID, Peripheral, constants
>>> uuid_env_sense = UUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e")
>>> serv_env_sense = Service(uuid_env_sense)
>>> periph = Peripheral()
>>> periph.addService(serv_env_sense)
>>> periph.advertise(device_name="micr_temp", services=[serv_env_sense])

 And I got feedback:
Traceback (most recent call last):
  File "<stdin>", in <module>
OSError: Can not apply advertisment data. status: 0x09

Rdgs,
Martin

linycm, 2018-11-11:
Maybe my example something wrong, I used old firmware version (should be softdevice3.0) as below, that work for bluetooth connection.

MicroPython v1.8.1-3865-gd91c351-dirty on 2018-05-01; PCA10040 with NRF52832

`from machine import RTCounter, Temp
from ubluepy import Service, Characteristic, UUID, Peripheral, constants
uuid_env_sense = UUID("0x181A")
serv_env_sense = Service(uuid_env_sense)
temp_props = Characteristic.PROP_NOTIFY | Characteristic.PROP_READ
temp_attrs = Characteristic.ATTR_CCCD
uuid_temp = UUID("0x2A6E")
char_temp = Characteristic(uuid_temp, props = temp_props, attrs = temp_attrs)
serv_env_sense.addCharacteristic(char_temp)
periph = Peripheral()
periph.addService(serv_env_sense)
periph.advertise(device_name="micr_temp", services=[serv_env_sense])`

  But when I changed to updated version with softdevice 6.0, bluetooth signal can be advertised but cannot connect.
  
Rdgs,
Martin

edwios, 2018-11-11:
I have the same problem, using the stock example ubluepy_temp.py, advertising is ok, event_handler EVT_GAP_CONNECTED ok, but then that's all. Connection soon timeout after a while.

glennrub, 2018-11-11:
@linycm, are you running this from a build of master branch? 

I'm not able to re-produce the error, as the code you posted advertise and connects for me. Do you have any log from the central that could hint on why the connection fails?

Alternatively, you could enable some logging from the ble_drv.c by toggling the BLE_DRIVER_VERBOSE to 1 to see why the connect might fail on the peripheral side.
https://github.com/micropython/micropython/blob/master/ports/nrf/drivers/bluetooth/ble_drv.c#L43

linycm, 2018-11-14:
using MicroPython v1.9.4 firmware, I try to use Android phone nRF connect APP, ubluepy_temp.py cannot work.  But if I put simple bluetooth code in REPL, it works for bluetooth connection. nRF connect APP ios version still cannot work.

glennrub, 2019-02-19:
@linycm , sorry for taking so long. Thanks a lot for your log!

Based on the log i do see an unhandled event, 0x23. This translates into BLE_GAP_EVT_DATA_LENGTH_UPDATE_REQUEST.

I do not have any way of provoking your error using iOS, so i have made a patch that you could try out:
https://github.com/glennrub/micropython/commit/4b57af5fdde3ec11aec6a0526133baaa9e89762e

Could you try out the patch and see if this solves the issue?

dhalbert, 2019-02-19:
We saw the same error in the CircuitPython BLE code. iPhone 7 and later, and modern iPads apparently have extended-length BLE packet capability in the hardware. iPhone 6s and earlier don't. We just say we don't provide this:
https://github.com/adafruit/circuitpython/blob/master/ports/nrf/common-hal/bleio/Peripheral.c#L269

linycm, 2019-02-27:
@glennrub , I compiled the new version again. Try again but I cannot connect BLE in iphone7 yet.

silbo, 2019-09-04:
@glennrub I tested it also on my nrf52840DK (PCA10056) board, it freezes when I try to connect to the device and the client device doesn't finish connecting to it (tested with Android, Mac and iOS)
it seems to get stuck at discovering services