micropython/micropython #4295: esp8266 - feature request, disable wifi radio

pacmac, 2018-11-07:
The esp8266 (compared to esp32, nrf etc) is by far the best supported micropython device with so much proven and tested modules and good amounts of flash and ram and well proven & tested.

It would be so good to be able to disable the wifi radio so that this can be used as a powerful microcontroller with 'acceptable' power consumption.

I have googled several times, but don't seem to be able to find a definitive answer as to whether it can be completely disabled, although there are several references in arduino and node forums to putting the radio to sleep.

```
WiFi.mode( WIFI_OFF );
WiFi.forceSleepBegin(); 
```

Is this actually possible ? - even if it was a build option to build with WIFI off would still be a huge benefit as there are many applications where the 8266 is perfect, except for it's power consumption using wifi.

I have spent the last 2 weeks getting familiar with the nrf devices, which are low power but when you compare the level of functionality as well as the ram, flash and filesystem, the 8266 would still be an ideal non-wifi device if only it can run without the radio on !

profra, 2018-11-07:
I vote by ten fingers for this option!!!

pacmac, 2018-11-08:
It would be sooo good if this could be added as a build option ?

profra, 2018-11-08:
This function should be functional for a long time in the esp library... or not?

[image]

profra, 2018-11-08:
[image]

pacmac, 2018-11-09:
@profra - I already tested the esp modem sleep a few weeks ago, it returns a message saying that the function is not implemented.

That would also be nice, to be able to put the modem to sleep programatically while having the rest of the mcu fully active, but being able to disable the radio at build time would be nice.

@dpgeorge - where can I add this code in my build folder ?

pacmac, 2018-11-09:
I just re-read your post again, and just noticed that you said at runtime and not at build.

I am not experienced with c and don't know how to call c functions from the micropython environment.

Would it be possible to elaborate a bit on how I would implement this so that I can test it ?

many Thanks

dpgeorge, 2018-11-09:
> Would it be possible to elaborate a bit on how I would implement this so that I can test it ?

You can try this patch, which will automatically enable sleep if both STA and AP are inactive (via `WLAN().active(0)` and `WLAN(1).active(0)`) and disable sleep if either interface becomes active again:
```diff
--- a/ports/esp8266/modnetwork.c
+++ b/ports/esp8266/modnetwork.c
@@ -83,7 +83,18 @@ STATIC mp_obj_t esp_active(size_t n_args, const mp_obj_t *args) {
         } else {
             mode &= ~mask;
         }
+        if (mode != NULL_MODE) {
+            wifi_fpm_do_wakeup();
+            wifi_fpm_close();
+            ets_loop_iter();
+        }
         error_check(wifi_set_opmode(mode), "Cannot update i/f status");
+        if (mode == NULL_MODE) {
+            wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
+            wifi_fpm_open();
+            wifi_fpm_do_sleep(0xfffffff);
+            ets_loop_iter();
+        }
         return mp_const_none;
     }
```

robert-hh, 2018-11-09:
No. It did not work in my trial attempt. The device simply reboots most of the time. And if not, the current is not changed. 
```
>>> WLAN().active(0)
force slp enable,type: 2
pm close 7
fpm open,type:2 0
fpm 758

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 30996, room 16 
tail 4
```

pacmac, 2018-11-09:
based on another thread where someone tried this in 2016, I created a new function:

```
STATIC mp_obj_t esp_force_modem_sleep() {
    wifi_station_disconnect();
    wifi_set_opmode(NULL_MODE);                             //set wifi mode to null mode.
    wifi_fpm_set_sleep_type(MODEM_SLEEP_T);                 //set force sleep type
    wifi_fpm_open();                                        //enable force sleep function
    return mp_obj_new_int(wifi_fpm_do_sleep(0xFFFFFFF));    //sleep until manual wakeup
}
```
This returns 0 which means the mode was set, however the current does not fall.

So apparently you have to set:
`#define MICROPY_REPL_EVENT_DRIVEN   (1)`

However, doing this causes build to fail:

```
CC esp_mphal.c
CC esp_init_data.c
CC gccollect.c
CC lexerstr32.c
CC uart.c
uart.c: In function 'uart_task_handler':
uart.c:291:30: error: 'input_buf' undeclared (first use in this function)
     while ((c = ringbuf_get(&input_buf)) >= 0) {
                              ^
uart.c:291:30: note: each undeclared identifier is reported only once for each function it appears in
../../py/mkrules.mk:47: recipe for target 'build/uart.o' failed
make: *** [build/uart.o] Error 1
```

pacmac, 2018-11-09:
I also tried the patch to esp_active, it ran successfully and the device did not reboot, but the power consumption did not change at all.

Is there a quick fix to prevent the error raised above during compile ?

profra, 2018-11-09:
@pacman I know that this function has not been working properly, I was only "a bit ironic". That is the reason why I have to use (against my will because I like MP) very often "arduino environment... WiFi.mode( WIFI_OFF); WiFi.forceSleepBegin();". Yes, this function is more important for those who are really working with physical computing, especially powered by batteries. It is not interesting for "pure SW humans". My knowledge of C and internal structure of MP is unfortunately weak to contribute to the solution.
My drawer is full of esp8266 and I look for solutions with batteries and minimal  power consumption. I have started the small project where I try to use simple and cheap microcontrollers from TI (MSP430FR2xxx) which will watch events around (potentially save data in the own ferromagnetic cells) and wake up ESP8266 only in the situations where it will be needed. I don't know another chip which would be so cheap, so accessible, so capable and so suitable for tinkering... and with MP :-).

@dpgeorge I would like to thank you at this point for a long-discussed topic on how to use UART (0) for other purposes than REPL. Your solution has opened a completely new space for applications with serial port needs. You solved it because you have a deep insight into the whole issue. It would be nice and it would be very appreciated by a lot of users if you were to become the "modem-sleep" solver. :-)

dpgeorge, 2018-11-13:
> I have done all the things that I can think of to try and get this working (see posts above) but have not succeeded.

Did you try compiling uPy with a more recent version of the ESP8266 SDK?  If the power saving works using the Arduino environment then it's simply a matter of software.

pacmac, 2018-11-13:
Damien, I am using the latest sdk from here:

`https://github.com/pfalcon/esp-open-sdk.git`

But are not sure if this is using a 

> more recent version of the ESP8266 SDK

But looking at the github site, it seems that the last update was a year ago.

So how can I update the SDK as the instructions I followed indicated that I should get the sdk as above ?

dpgeorge, 2018-11-13:
> So how can I update the SDK as the instructions I followed indicated that I should get the sdk as above ?

You can try one (or both) of these PRs for esp-open-sdk:
https://github.com/pfalcon/esp-open-sdk/pull/332 (SDK v2.2.1)
https://github.com/pfalcon/esp-open-sdk/pull/344 (SDK v3.0.0)

(To avoid long build times of the toolchain it may be enough to do `make clean-sdk clean-sysroot all` to build each PR and then test them.)

pacmac, 2018-11-13:
Thanks, and sorry for my ignorance but is this the correct procedure ?:

```
  git fetch origin pull/344/head
  git commit
  git checkout -b pullrequest FETCH_HEAD
  git submodule update --init
  make clean-sdk clean-sysroot all
```
Thanks

pacmac, 2018-11-13:
Unfortunately, the build-clean of 8266 fails:

```
.....
CC ../../py/mpprint.c
In file included from ./esp_mphal.h:76:0,
                 from ../../py/mphal.h:32,
                 from ../../py/mpprint.c:33:
./etshal.h:9:6: error: conflicting types for 'ets_delay_us'
 void ets_delay_us(uint16_t us);
      ^
In file included from ./esp_mphal.h:69:0,
                 from ../../py/mphal.h:32,
                 from ../../py/mpprint.c:33:
/nfs/qnap/data/uPython/build/micropython/ports/esp8266/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/include/osapi.h:33:6: note: previous declaration of 'ets_delay_us' was here
 void ets_delay_us(uint32_t us);
      ^
../../py/mkrules.mk:47: recipe for target 'build/py/mpprint.o' failed
make: *** [build/py/mpprint.o] Error 1
```

pacmac, 2018-11-14:
It seems there is a note about adding user_pre_init()

https://github.com/espressif/ESP8266_NONOS_SDK

Where should I add this ??

pacmac, 2018-11-14:
OK, solved, now it builds.

pacmac, 2018-11-14:
OK, am now able to build, but after flashing the device will not boot :-(

**1) esthal.h**
`// void ets_delay_us(uint16_t us);`

**2) posix_helpers.c**
`/*
void *malloc(size_t size) {
    void *p = gc_alloc(size, false);
    if (p == NULL) {
        // POSIX requires ENOMEM to be set in case of error
        errno = ENOMEM;
    }
    return p;
}
void free(void *ptr) {
    gc_free(ptr);
}
void *calloc(size_t nmemb, size_t size) {
    return malloc(nmemb * size);
}
void *realloc(void *ptr, size_t size) {
    void *p = gc_realloc(ptr, size, true);
    if (p == NULL) {
        // POSIX requires ENOMEM to be set in case of error
        errno = ENOMEM;
    }
    return p;
}
*/`

**3) main.c**
`void user_pre_init(void){
  
}`

making these 3 changes and the build completed, however the device will not boot.

pacmac, 2018-11-14:
so it seems that user_pre_init() needs to call system_partition_table_regist(), but I have no idea what params I need to send !

```
void user_pre_init(void){
  system_partition_table_regist();  
}
```

pacmac, 2018-11-14:
Hi Damien, 

Any suggestions on how to implement system_partition_table_regist() ?

dpgeorge, 2018-11-15:
I checked with esp8266 Arduino and modem sleep works.  Specifically, putting this in `setup()` drops current consumption by about 50mA:
```c
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);
  wifi_fpm_open();
  wifi_fpm_do_sleep(0xffffffff);
```
The Arduino environment seems to use the esp8266 non-os SDK 2.2.0-28.

---

I tried non-os SDK 2.2.0 and 2.2.1 with uPy esp8266 and neither of them give reduced current with the above lines of C code.  So there must be something else that is needed.  (Note: to get 2.2.0 and 2.2.1 working required a `make erase` before `make deploy`.)

pacmac, 2018-11-15:
Thanks, I have been trying to use SDK version 3.0, however V3 requires (in main.c):

```
void user_pre_init(void){
  system_partition_table_regist();  
}
```

https://github.com/espressif/ESP8266_NONOS_SDK

And I have no idea what params to pass.

dirkmueller, 2018-11-25:
@dpgeorge the nonos sdk documentation examples also call wifi_set_opmode(..NONE..) before the sleep setting, so turning off wifi mode is maybe needed? Or at least yielding?

https://github.com/esp8266/Arduino/blob/79a6f36a029055c36f1a400742a4f26daf86e459/libraries/ESP8266WiFi/src/ESP8266WiFiGeneric.cpp#L396

It looks like there is also sleep_level which can be increased over the minimal setting which is default

dpgeorge, 2018-11-26:
> Yes I got esp8266 SDK 3.0.0 working.

See #4318 for minimal changes required to get SDK 3.0.0 booting.

> the nonos sdk documentation examples also call wifi_set_opmode(..NONE..) before the sleep setting, so turning off wifi mode is maybe needed? Or at least yielding?

Yes, I did make sure that the wifi opmode was NULL_MODE.  Note that yielding is different for the esp8266 implementation here, compared with Arduino, it's done automatically by the VM (or can be explicit via a call to `ets_loop_iter`).

dpgeorge, 2018-11-28:
The problem was that the ets idle callback was not being called, so the radio was not being put to sleep.  A PR for this radio sleep feature is now available in #4320

dpgeorge, 2018-12-04:
This feature was implemented in 321d75e08769b82d951fdf65379dab8d0ce76e2f and 62b4bebf6437df1a1a82747421c1ab94f46c15c2

closed 2018-12-04 by dpgeorge