micropython/micropython #4296: I found some err from py/vm.c:66 line

xiaoxiaohuixxh, 2018-11-08:
I found some err from py/vm.c:66 line
#define DECODE_PTR \
    DECODE_UINT; \
    void *ptr = (void*)(uintptr_t)code_state->fun_bc->const_table[unum]
it cant run well when i use 64bit platform.
i changed it to 
void *ptr = (mp_raw_code_t *)(*((mp_raw_code_t **)((uint8_t *)((void*)(uintptr_t)code_state->fun_bc->const_table) + (8 * unum))))
it run well.

dpgeorge, 2018-11-08:
> I found some err from py/vm.c:66 line ...
> it cant run well when i use 64bit platform.

What lead you to find an error here, what were the symptoms (the problems) that came up?

The VM is known to work well with 64-bit platforms like x86-64.  What 64-bit platform do you use, is the typedef for `mp_int_t` (and mp_`uint_t`) defined correctly?

xiaoxiaohuixxh, 2018-11-08:
Do you use the uintptr_t that in the stdint.h?i use the kendryte 210 mcu.

xiaoxiaohuixxh, 2018-11-08:
ptr is zero when the unum > 0

dpgeorge, 2018-11-09:
> Do you use the uintptr_t that in the stdint.h?

Yes `uintptr_t` should be defined by stdint.h, and it should be the same size as a pointer word.

> i found point only right shift 32 bit when the unum add 。

Can you elaborate on what exactly you mean by this?

You should also check the sizes of the following types for your platform: `sizeof(void*)`, `sizeof(uintptr_t)`, `sizeof(mp_int_t)`, `sizeof(mp_uint_t)`.  Eg print out those values when the code starts up.  For a 64-bit platform they should all be 8 bytes big.

dpgeorge, 2018-11-26:
Assuming this was fixed in the 3rd party port.  Feel free to reopen with more details if errors persist.

closed 2018-11-26 by dpgeorge