micropython/micropython #4297: upip is failing with esp8266

gyldcs, 2018-11-08:
same issue as #3743

after trying to install a package I get "Error installing '____': , packages may be partially installed"
using the latest stable build, that should have the fix, so I do not know what is the issue

the url seems fine:
https://pypi.org/pypi/micropython-json/json

pfalcon, 2018-11-08:
> I get "Error installing '____': , packages may be partially installed"

If you get that, it means that you tried to installed package named "____". There's no such package on PyPI, so you get the error rightfully.

gyldcs, 2018-11-08:
well, it's on pypi, so `upip.install("micropython-json")` should work. 
I've added "____" as it does not work with any package at all

peterhinch, 2018-11-09:
I get the following failure here on the reference board. Only **decoder.py** is installed.
```python
MicroPython v1.9.4-684-g1dd4d93-dirty on 2018-11-06; ESP module with ESP8266
Type "help()" for more information.
>>> 
>>> upip.install("micropython-json")
Installing to: /lib/
Warning: pypi.org SSL certificate is not validated
Installing micropython-json 0.1 from https://files.pythonhosted.org/packages/3a/b3/6b5dc8008c66ec2e6e1ccd1554f3ebffdc8dd4e872bd1f9f276921f43f7f/micropython-json-0.1.tar.gz
Error installing 'micropython-json': [Errno 22] EINVAL, packages may be partially installed
>>> 
```

pfalcon, 2018-11-10:
> [Errno 22] EINVAL, packages may be partially installed

Thanks for the report. This may be a sign of a package which wasn't gzipped with 4K dictionary, and thus would have troubles being decompressed on a low-heap system like esp8266. Indeed, that was the case. A new version was uploaded, would be nice if you could verify.

peterhinch, 2018-11-12:
The outcome is now
```python
MicroPython v1.9.4-684-g1dd4d93-dirty on 2018-11-06; ESP module with ESP8266
Type "help()" for more information.
>>> import upip
>>> upip.install("micropython-json")
Installing to: /lib/
Warning: pypi.org SSL certificate is not validated
Installing micropython-json 3.3.3.post1 from https://files.pythonhosted.org/packages/15/23/cd83249933dc5be819f286adf8b2f0cbf63cb382ff1e87a75ada8b2a3727/micropython-json-3.3.3.post1.tar.gz
LmacRxBlk:1
Error installing 'micropython-re-pcre': , packages may be partially installed
>>>  
```
The five files are installed to the target but it seems to have fallen over installing a dependency: there is no sign of **micropython-re-pcre** there.

peterhinch, 2018-11-13:
I've now repeated the test as follows.

I deleted the lib directory. Then I ran the installation three times, issuing a soft reset before each attempt. The outcome was the same: the five json files were installed and no others.

It's also worth noting that on the last run I saw the infamous **LmacRxBlk:1** (I soft-reobooted each time to minimise the likelihood of this).

pfalcon, 2018-11-13:
> It's also worth noting that on the last run I saw the infamous LmacRxBlk:1

That's actually what I meant, I wondered if you hinted that some run of upip guaranteedly leads to it, or it just nondeterministically appears, which is a well known fact.

I assume that you were getting `Error installing 'micropython-re-pcre': , packages` error each time. Then it must be a MemoryError. In https://github.com/pfalcon/micropython-lib/commit/8d87934abb0795e609ea7d3ba68d8e4891926789 , I fixed error reporting, getting back to old good tracebacks away from fairly misguided "users should not know details of the errors".

Otherwise, upip needs further memory usage optimizations, that's also a well known fact.

And for avoidance of doubt, let me mention another obvious (?) fact - there's totally no sense in installing micropython-json on esp8266, it won't work there. (For me it's a case of "let's push upip to extreme on hw like esp8266 and see how exactly it breaks".)

peterhinch, 2018-11-13:
It's not deterministic and I do get the `Error installing 'micropython-re-pcre': , packages` message every time. This strikes me as odd. If I install with `upip` under Unix it runs to completion with no error message. It installs the five files and does not install `micropython-re-pcre`. So it's not clear to me why, on the ESP8266, it produces that particular message.

I take your point about the package size and the likelihood of OOM on the ESP8266 - so maybe the issue can be closed.

If there's any further testing I can do, I'm happy to help.

pfalcon, 2018-11-13:
> This strikes me as odd.

If it does, then you didn't read what I wrote above carefully enough.

> If I install with upip under Unix it runs to completion with no error message.

Of course, there's much more memory. Run unix port with similarly limited heap and you'll get the "same" error in no time. It will be the same, modulo detailedness of the exception message (on esp8266 not detailed).

> It installs the five files and does not install micropython-re-pcre.

Impossible, because micropython-json's dependencies clearly specify micropython-re-pcre.

> If there's any further testing I can do, I'm happy to help.

Not now, thanks. Of course, per your likes, you instead can go over the information above to understand well what happens, and to confirm the fact that any systematic (vs nondeterministic) behavior happening on esp8266 can also be reproduced on the unix port.

iBobik, 2018-11-17:
Is it possible to download packages in plaintext? Micropython packages are small enought it will not be problem and we can save resources by no need to decompress it.

peterhinch, 2018-11-18:
You can download from PyPi to a PC in two ways. One is to use **upip** under the Unix build. If you don't want to install the Unix build, **micropip** does the same job under CPython 3.2 or later. It can be found in [this repo](https://github.com/peterhinch/micropython-samples).

Alternatively you can examine the files in micropython-lib.

srohit0, 2019-12-12:
We are trying to port **[deepC](https://github.com/ai-techsystems/dnnCompiler)**,  machine learning inference framework and compiler at https://github.com/ai-techsystems/dnnCompiler to micropython, and getting similar error.

```code
./micropython -m upip install deepC
```

> Installing to: /home/amd/.micropython/lib/
> Warning: micropython.org SSL certificate is not validated
> Installing deepC 0.121 from https://files.pythonhosted.org/packages/4b/32/21103ba9ce471d96c12224b688571ee70dfde0055a4381848275dc64fec8/deepC-0.121-cp37-cp37m-linux_armv7l.whl
> Error installing 'deepC': compression header, packages may be partially installed

---

Likely, there is something at our end we need to do. But this error has no information to continue making progress.

Any help/experience-share/workaround will be helpful.

IveJ, 2019-12-12:
Hi,

Let try another way to install step by step.

QGB, 2021-02-14:
Error: Unable to resolve micropython.org (no Internet?)

fepegar, 2021-08-23:
I solved this with a soft reset (Ctrl + D).

srohit0, 2021-08-28:
You mean you were successful in installing [deepC](https://github.com/ai-techsystems/deepC) on to micropython using upip under the Unix build?

fepegar, 2021-08-28:
No, I was trying to install something else on an ESP32 using `upip` and was getting the error message: [...] `packages may be partially installed`.

jonnor, 2024-10-01:
upip has been replaced by mip in MicroPython 1.20, https://github.com/micropython/micropython/releases/tag/v1.20.0
Closing as no longer relevant.

closed 2024-10-01 by jonnor