micropython/micropython #4298: esp8266 stops processing wifi when (ab)using micropython.schedule()

pmp-p, 2018-11-09:
use case was scheduling asyncio loop step while REPL is running like it can be done with cpython PyOS_InputHook ( cf https://gist.github.com/pmp-p/2cc65bc914cb046972a3e94003fe7f1c )

```
def hog(v):
    Time.sleep_us(1) # <= will make wifi hang
    micropython.schedule( hog , 1)

micropython.schedule( hog , 1)
```

dpgeorge, 2018-11-15:
Can you please clarify what is meant by "stops processing wifi" / "will make wifi hang"?  The above "hog" function will not let anything else run so it's not surprising that wifi does nothing.

Also, if the `sleep_us()` time is increased, does that fix things?

pmp-p, 2018-11-15:
not fixing, i think chaining scheduling calls steps over the wifi processing. i guess if scheduling is started/stopped fast from the repl then it looks like wifi processing resuming. please could you point me where is the mandatory call path for processing wifi from within repl loop , i 'll try to get around  ?

dpgeorge, 2018-11-16:
I tried to reproduce this but I cannot.  I enabled the STA and AP interfaces of the esp8266, then connected the STA to the local wifi network.  Then I ran:
```python
>>> def f(x):
...     time.sleep_us(1)
...     micropython.schedule(f, x)
... 
>>> import time, micropython
>>> micropython.schedule(f, 1)
>>> micropython.schedule(f, 2)
>>> micropython.schedule(f, 3)
>>> micropython.schedule(f, 4)
>>> micropython.schedule(f, 5)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: schedule stack full
```
Which should definitely hog the CPU with 4 simultaneous instances of `f` scheduled at once (and rescheduling each other).  With this code running a PC could still ping the esp8266 over the STA interface without any packet loss and without any noticeable difference in ping time.  Also, another device could connect to the esp8266's AP while all this was going on.

peterhinch, 2018-11-16:
I can confirm this. Nor does the following stop the reference board from responding to pings:
```python
while True:
   pass
```

pmp-p, 2018-11-16:
i'm lost as i pasted your test in boot.py 
```
def f(x):
    time.sleep_us(1)
    micropython.schedule(f, x)

import time, micropython
micropython.schedule(f, 1)
micropython.schedule(f, 2)
micropython.schedule(f, 3)
```
i can't even connect to my AP called "AP"
uart gets spammed with 
```
no AP found, reconnect after 1s
reconnect
```
or when doing it from repl 
```
bcn_timout,ap_probe_send_start
ap_probe_send over, rest wifi status to disassoc
state: 5 -> 0 (1)
rm 0
```

i'll retry with another board asap.
when i made the report i had the code in main.py

pmp-p, 2018-11-16:
seem the same board works fine using milliseconds, but such values are not usefull.
```
def f(x):
    time.sleep(.01)
    micropython.schedule(f, x)

import time, micropython
micropython.schedule(f, 1)
micropython.schedule(f, 2)
micropython.schedule(f, 3)
```
so i guess it's a design/quality problem

pmp-p, 2018-11-16:
> Also, if the `sleep_us()` time is increased, does that fix things?

@dpgeorge very sorry for being misleading, i indeed did not increase enough time to see if it would fix things a bit i did not expect such high values
2500 is the fastest best i could get
```
def f(x):
    #BAD 2000 2250 2350 2400
    # variable results ampy can fail 2450
    #OK 2500 
    time.sleep_us(2500)
    micropython.schedule(f, x)

import time, micropython
micropython.schedule(f, 1)
#micropython.schedule(f, 2)
#micropython.schedule(f, 3)

```
this a picture of the board , filename says it all https://github.com/pmp-p/stupyde/blob/master/doc/not_weemos.png

peterhinch, 2018-11-16:
Code should not normally be put in `boot.py`. For code which is to auto-run I normally have something like this in `main.py`:
```python
import foo
foo.run()
```
My practice is to have `run()` busy-wait until the ESP8266 has connected, and only then start doing useful work. If your code was competing with the ESP8266 attempting to connect that might explain why your results differ from ours: I pasted code at the REPL after the board had connected.

pmp-p, 2018-11-16:
@peterhinch i also tried to put the sample in a module truc.py and then "import truc.py" long after board is re-connected ~ 10 secondes. i got same time limit   2500,   (2450 is unreliable , eg board still ping but ampy won't be able to transfer a file )

i'll try to grab a old lolin board tomorrow

pmp-p, 2018-11-16:
well just had to tidy the drawers a bit , here's my findings :

fail < 2500 µs
wemos? https://github.com/pmp-p/stupyde/blob/master/doc/maybe_wemos.png
nomarkings: https://github.com/pmp-p/stupyde/blob/master/doc/not_weemos.png
both failing boards can be found ~ 1.7 €

Works same as dpgeorge reference board, ie as expected with 1µs delay , from boot/main or repl:
https://github.com/pmp-p/stupyde/blob/master/doc/nodemcuv3.png
~ 2.5€

peterhinch, 2018-11-17:
This would seem to confirm my own findings: some ESP8266 boards are rubbish. I'm sure you'll agree that unless a problem can be replicated on the reference board it can't be considered a bug.

I wonder if your "wemos?" board is actually from WeMos. Their D1 mini is good (in my experience).

pfalcon, 2018-11-17:
> some ESP8266 boards are rubbish

Different esp8266 implementations definitely differ widely. While working on the port, I had a cases when some code crashed with exception on one board (pretty deterministically, i.e. within short period of time), which worked fine for extended time on another. Power cabling was fine, though it's always possible to blame components on a particular boards.

But then I had countless times when dreaded "LmacRxBlk:1" popped up on one board (i.e. non-deterministically), while behaved well on another board (i.e. never popped up out of sudden; otherwise, it's well-known that one can induce "LmacRxBlk:1" deterministically, by not reading incoming data/not closing sockets).

All that lead me to conclusion that the only way forward is to completely reverse-engineer wifi side of things and write open-source implementation of it, and that's where it's at for me.

pmp-p, 2018-11-17:
yeah that's the game of try cheaper stuff, sadly the cheap ones mentionned have breadboard friendly layout guraranted, and far better vernish than random lolin . that is regardless of the price.

Any way to detect a board or another at boot via python and official firmware ? so i would set the timer accordingly for repl ?  2.5 ms is not that bad for people starting with async and blinking leds.

pmp-p, 2018-11-17:
> I'm sure you'll agree that unless a problem can be replicated on the reference board it can't be considered a bug.
well i don't, as i tend to make a difference between reference and pratical useabality, favouring the latter.
For me useable board is the most recent, cheapest and widely avalaible design, shiny fresh out of the oven and not discontinued items picked up from a rusty stash. and with respect to people having not much money for learning and could only afford 1 rubbish model they would certainly considerer that as an annoyance if not a bug.
but as pfalcon suggested the bug *may* be deeper down than micropython firmware and it's certainly not good news.

peterhinch, 2018-11-18:
@pmp-p The [reference board](https://www.adafruit.com/product/2821) is readily available. Sure, it costs more. Whether cheap boards are worth buying depends on whether you place a value on your time.

@pfalcon I spent a great deal of time trying (and failing) to get reliable operation from Sonoff modules. I concluded there was a design problem with the hardware, although I couldn't identify it either by measurement or by studying the schematics. However WiFi boards have relatively powerful radio emissions close to signal lines: there is scope for problems which would require sophisticated equipment to study.

> All that lead me to conclusion that the only way forward is to completely reverse-engineer wifi side of things and write open-source implementation of it, and that's where it's at for me.

That would be awesome, but I wonder if the effort would better be directed at ESP32, which has its own problems.

pmp-p, 2018-11-18:
haha @peterhinch , pardon me i believed that this precise board was made for Adafruit fans and/or Circuitpython. Meanwhile i hope reference OS for cpython doesn't turn to win-something on EFI hardware because GNU/Linux is so cheap on rubbish on sub-$8 arm boards.  ;)

Sarcams apart, the chip on the weird cheap board looks  *a lot* like the ones on SonOff modules. did you have a close look at SonOff markings ?

pfalcon, 2018-11-18:
> pardon me i believed that this precise board was made for Adafruit fans and/or Circuitpython

Well, to confirm @peterhinch's words, Adafruit ESP8266 Feather (or how it's called) was the reference/recommended MicroPython ESP8266 board during the Kickstarter campaign, and remains such (as nobody followed the process to change that, and why would they?). This was done by exactly the reasons discussed here - it was well known that ESP8266 boards vary widely in quality (and it was only worse back then), so we wanted to "recommend" users a vendor which could be trusted, in particular, a vendor which could be reached in case of hardware-related problems. Btw, I believe there wasn't CircuitPython at the time.

I also fully agree with @peterhinch re: cheap boards vs price of own time. I'd even say that cheap boards are only useful if you want to participate in motion "to change the world", as mostly happened with ESP8266. I for one don't spend much of my time on ESP32 because I don't see it as revolutionary as ESP8266, but actually boring, following the suit of chips (or modules) from TI, Broadcom/Cypress, etc. But those guys have documentation, quality, and (almost) top-notch history of support. Again, ESP32 now "almost" has the same, or more specifically, they strive to be like they, which again, mostly makes it a boring copycat (aka: we won as a community, and now the vendor does its work instead of us, and we judge by the results, and heck, there's a lot of vendors on the plate).

pmp-p, 2018-11-18:
@pfalcon  thanks for clarifying i went very late after the campaign.
Maybe i'm dumb but i'm reluctant to advise throwing money at fabless people promoting software forks.

So as i would like equal rights for all boards ( that's my "change the world" vision I promote for H3 cpu based boards  too ) any official way to distinguish boards from micropython and set timers accordingly ( and still keep max speed if wifi not used ) ?. turnining the bug into a feature seems more easy than buying *all* models.
 If there's not such board identification method i'm willing to write a self-test but that would save me time.

pfalcon, 2018-11-18:
> Maybe i'm dumb but i'm reluctant to advise throwing money at fabless people promoting software forks.

Well, I'd say, there's no such advise. (And I have my share of criticism of Adadruit. Let me just say that I'm happy they have a fork, would be very depressing to see all their stuff in the mainline.)

As I see it, the various parties effectively hint the following: if there's an issue you report, and it was investigated by various parties as occurring on one hardware and not on another, then the best response the project can give is: "Please contact the vendor of your board regarding this issue."

> any official way to distinguish boards from micropython

There definitely should be. When you start/reboot MicroPython, it prints the name of the board in REPL header. So there got to be the way. At this point, I'd turn to the documentation, because honestly, I don't remember the exact details, and would need to search them before giving answer to you, but I hope you can do it yourself.

If you can't find anything, tell us where/what you tried, so we can see if it needs improvement, and we'll look into it. (By "we" I mean "somebody of people lurking thru tickets, who has time in a particular moment").

pmp-p, 2018-11-19:
found (blindly) a fix that works 99.99% of time i hope it will help someone clarify what happens really
https://github.com/micropython/micropython/compare/master...pmp-p:patch-2

pmp-p, 2019-06-24:
Also dhcp address negociation requires more OS servicing cycles than normal usage. On the long run even weemos D1mini will loose connectivity if renewal happens under load. 

condition of test linux isc-dhcpd server to have decent logs, bridged on a very common tplink wifi router with dhcpd off.