micropython/micropython #4299: py/modio.c Can this be made to work in Unix build?

peterhinch, 2018-11-10:
The ability to subclass **io.IOBase** is unavailable under Unix. It would be good if this were remedied.

dpgeorge, 2018-11-15:
> The ability to subclass **io.IOBase** is unavailable under Unix.

I'm pretty sure it's enabled.  With the unix build one can do this:
```
$ ./micropython
MicroPython v1.9.4-684-g51482ba92-dirty on 2018-11-15; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import uio
>>> uio.IOBase
<class 'IOBase'>
>>> class A(uio.IOBase):
...     def write(self, data):
...         print('wr', data)
... 
>>> print('a')
a
>>> print('a', file=A())
wr bytearray(b'a')
wr bytearray(b'\n')
```

peterhinch, 2018-11-15:
With Unix port built from source:
```
MicroPython v1.9.4-683-gd94aa57-dirty on 2018-11-15; linux version
Use Ctrl-D to exit, Ctrl-E for paste mode
>>> import io
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'io'
>>> 
```
To fix this I modified unix/mpconfigport.h as follows. Note I am far from confident that these changes are correct.
```
#define MICROPY_PY_IO               (1)
```
also
```
#define MICROPY_PORT_BUILTIN_MODULES \
    MICROPY_PY_FFI_DEF \
(lines omitted)
    { MP_ROM_QSTR(MP_QSTR_io), MP_ROM_PTR(&mp_module_io) }, \
```
I could now import the module but my attempts to create StreamReader instance failed. I used this demo class, proven on the Pyboard:
```python
import uasyncio as asyncio
import utime
import io
MP_STREAM_POLL_RD = const(1)
MP_STREAM_POLL = const(3)
MP_STREAM_ERROR = const(-1)

class MillisecTimer(io.IOBase):
    def __init__(self):
        self.end = 0
        self.sreader = asyncio.StreamReader(self)

    def __iter__(self):
        await self.sreader.readline()

    def __call__(self, ms):
        self.end = utime.ticks_add(utime.ticks_ms(), ms)
        return self

    def readline(self):
        return b'\n'

    def ioctl(self, req, arg):
        ret = MP_STREAM_ERROR
        if req == MP_STREAM_POLL:
            ret = 0
            if arg & MP_STREAM_POLL_RD:
                if utime.ticks_diff(utime.ticks_ms(), self.end) >= 0:
                    ret |= MP_STREAM_POLL_RD
        return ret

async def test():
    timer = MillisecTimer()
    for _ in range(10):
        await timer(300)
        print('test')

loop = asyncio.get_event_loop()
loop.run_until_complete(test())
```

dpgeorge, 2018-11-15:
> >>> import io

Ok, so the issue is that it the `io` module does not exist, but the `uio` module does.  Really there should be weak links for these modules (like on bare metal ports) so that one doesn't need to install `micropython-lib` to get basic compatibility when importing non-u modules.

> #define MICROPY_PY_IO (1)

That's not needed, it is enabled by default in py/mpconfig.h.

> I could now import the module but my attempts to create StreamReader instance failed

How exactly did it fail?  io.IOBase does exist.

pfalcon, 2018-11-15:
> Really there should be weak links for these modules (like on bare metal ports)

There's another opinion, that baremetal ports should drop those adhoc "weak links" which has become a mis-convenience. People should understand what is the target of the code they write - MicroPython or CPython, and use the corresponding module. (For CPython compatibility, get a compatibility module first.)

peterhinch, 2018-11-15:
Thanks for the pointer. I'll debug the code and raise a new issue when I can accurately describe how and why it fails under Unix.

closed 2018-11-15 by peterhinch

peterhinch, 2018-11-15:
See #4304