micropython/micropython #4300: ubluepy_temp.py stopped working on nRF52 s132 6.0.0

edwios, 2018-11-11:
I am not exactly sure if this is the same as issue #4288 but if it is, then I apologise for the duplication.

The stock example [ubluepy_temp.py](https://github.com/micropython/micropython/blob/master/ports/nrf/examples/ubluepy_temp.py) on a nRF52-DK PCA10040 board is no longer working after the SoftDevice is upgraded to s132 6.0.0 together with the master branch version of the micropython.

Advertising is ok, I can see it properly on the nRF connect. The event handler got EVT_GAP_CONNECTED and things got executed correctly. Then that's it. There is no further happenings between the nRF connect and the PCA10040. The status on the nRF connect Services page shows Connecting. Where the Services selections should be shown is now only a spinning cursor on a white page.

What was changed? Any idea how to fix this?

glennrub, 2018-11-11:
@edwios , i tried the ubluepy_temp.py as a frozen module just now, and i do not see any issue with it. 
Did you enable the notifications by writing the CCCD?

For reference, i used a pca10040 and an Android phone with "nRF Connect" application. Next, after connect, i enabled the services by writing the CCCD's, and then the notifications started to come.

edwios, 2018-11-30:
Some updates here. I've tried it on a brand new 52840-DK board with s140 6.0.0, same result. However, if I use the nRF Connect on my **computer**, it works perfectly. I can see the service and chars and read the temperature and set CCCD for notifications.

So it is only when I am trying to connect via iOS (both LightBlue and nRF Connect) that the connection doesn't work (don't know about Android, don't have one). Any idea?

_Note the apps have no issue connecting to other BLE devices as you can see on the attached screen recording_.

[ScreenRecording_11-30-2018 14-15-26.mov.zip](https://github.com/micropython/micropython/files/2632268/ScreenRecording_11-30-2018.14-15-26.mov.zip)

glennrub, 2019-02-11:
@edwios, have you got any further on this? 

I got the time now to try out the ublupy_temp.py on my old iphone5 using the "nRF Connect". I'm not sure how old this app is, but its of version 1.7.5. It seems to work OK for me there. I did however see i have a pending iOS upgrade on hold. I'll try to get that one in, and see how it performs.

projectgus, 2024-11-20:
The S130 SoftDevice version is 6.1.1 now, so at the very least this would need re-confirming against the newer version.

If someone bumps into this in the future then please open a new issue with the details.

closed 2024-11-20 by projectgus