micropython/micropython #4303: ESP32  _thread don't exit

Derrick45, 2018-11-14:
I write a test.py ,Mainly used to test thread.  as follows: 
import _thread
import time 
    
def func1():
    while 1:
        print("this is func1")
        time.sleep(1)
    
def func2():
    while 1:
        print("this is func2")
        time.sleep(1)

_thread.start_new_thread(func1, ())
_thread.start_new_thread(func2, ())

Then i send  the file to micropython filesysterm.Enter "import test" in interactive mode.I can see the code is running. But I can't stop it by pressing CTRL +c or CTRL +d.I wonder what other way to stop a micropython multithreaded program than to hard reset it.

robert-hh, 2018-11-14:
Besides the fact, that this question is a good topic for the forum, I had that problem too. A kill thread fucntion is missing. So on way would be to have the thread looking itself for a  stop signal, like the valua of a global symbol.

dpgeorge, 2018-11-15:
No, there is no way to force-kill a thread via an external mechanism, and it's not recommended to do this, see eg https://stackoverflow.com/questions/323972/is-there-any-way-to-kill-a-thread

> But I can't stop it by pressing CTRL +c or CTRL +d

Soft reset (crtl-d) is supposed to kill all threads (since the uPy heap is reinitialised on soft reset it doesn't make sense to keep threads around).  But this is currently not working and is discussed a bit more in #3980.  I'll close this issue because the bug is tracked by that issue.

closed 2018-11-15 by dpgeorge