micropython/micropython #4311: Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.

qpepfgh1, 2018-11-20:
What is the cause?

Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Core 0 register dump:
PC      : 0x400d199c  PS      : 0x00060830  A0      : 0x00000000  A1      : 0x3ffbf040  
A2      : 0x00000000  A3      : 0x3ffc3e15  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800d199c  A9      : 0x3ffbf020  
A10     : 0x00001d2e  A11     : 0x00000001  A12     : 0x3ffc1360  A13     : 0x3ffbf110  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400d199c:0x3ffbf040 0x7ffffffd:0x3ffbf090

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310

peterhinch, 2018-11-20:
Two comments. This should initially be raised in [the forum](https://forum.micropython.org/): as it stands it really doesn't constitute a usable bug report, and other users may be able to help you with it.

Secondly, nobody will be able to help unless you provide a lot more details. I suggest you read [how to submit a bug report](https://www.chiark.greenend.org.uk/~sgtatham/bugs.html).

qpepfgh1, 2018-11-20:
Thank you for a good point!
I am using FreeRTOS with the esp32 WROOM32 in the Arduino IDE.
I am trying to create about 4 threads using the xTaskCreate API. I get an error.
EX) xTaskCreate (wifiSet, "WiFi", 4000, NULL, 10, NULL);
          xTaskCreate (mqttSet, "MQTT", 4000, NULL, 10, NULL);
          xTaskCreate (signState, "sign", 4000, NULL, 10, NULL);
          xTaskCreate (HCState, "HC", 4000, NULL, 10, NULL);

Please tell us if you want to see the full code!

MikeTeachman, 2018-11-20:
For Arduino/ESP32 development I think you will get the best response by posting in the ESP32 Arduino forum:   
https://www.esp32.com/viewforum.php?f=19

or the github repo for the Arduino ESP32 core:
https://github.com/espressif/arduino-esp32

closed 2018-11-26 by dpgeorge