micropython/micropython #4313: Touch Pad Error

Fuermohao, 2018-11-21:
Touch pad error
```python

>>> import machine
>>> from machine import Pin
>>> t = machine.TouchPad(Pin(32))
>>> t.read()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: Touch pad error
```

_Originally posted by @Fuermohao in https://github.com/micropython/micropython/pull/3531#issuecomment-440129205_

Fuermohao, 2018-11-21:
my board is NodeMCU-32S.
[image]

nickzoic, 2018-11-21:
Thanks!  If you can please also add in details of which touch channels (all of them? only some of them?) are affected that'd help a lot.
`ports/esp32/machine_touchpad.c:mtp_read()` returns that error for any return from the ESP-IDF function `touch_pad_read` other than ESP_OK ... that might occur if the pin is not oscillating ...  Is anything else attached or is this reproducible on just the bare board?

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripheral/touch_pad.html#_CPPv214touch_pad_read11touch_pad_tP8uint16_t

Fuermohao, 2018-11-21:
@nickzoic 
> If you can please also add in details of which touch channels (all of them? only some of them?) are affected that'd help a lot.
ok, but I still need some time to figure out.

Fuermohao, 2018-11-21:
@nickzoic 
 I only found this pin layout. As for schematic I didn't find it.

Fuermohao, 2018-11-21:
[image]

@nickzoic  I tried this @MrSurly  
It seems a little different from yesterday.

Further, I found that when I touch the touchpad, an error is triggered and it will not be released.
[image]

Fuermohao, 2018-11-21:
[image]
here is my breadboard wiring diagram

nickzoic, 2018-11-21:
Oh!  What's the little red module?
If it's one of these kind of things
https://sktechworks.ca/product/tp223-capacitive-touch-module/ you don't
actually need it, the ESP32 has that function built in ...
Try removing the module and just touching the yellow wire!

Alternatively, if you want to use the module for some other reason, you
can treat the GPIO32 as a normal logic input pin.

Fuermohao, 2018-11-21:
oh！I got it... I am so stupid...
I removed it, now GPIO32 works fine.
So let's just shutdown this issue.

closed 2018-11-21 by Fuermohao

nickzoic, 2018-11-21:
I'm glad we were able to help!
Have fun!