micropython/micropython #4314: esp8266: UART(0) activation for REPL even if expressly deactivated

ARF1, 2018-11-21:
I am not sure if this is a bug of a feature. Before submitting a PR I thought I would ask...

Assume the following situation:

- esp8266 port
- webrepl activated
- `uos.dupterm(machine.UART(0, 115200), 1)` in `boot.py` commented out

In this situation, on boot `UART(0)` is still activated. This is fairly counter-intuitive since the user took extra pains to disable `UART(0)` for REPL. - Or so s/he thought...

Is this intentional?

The "problem" stems from the fact that `mp_reset` checks whether after running `boot.py` any `dupterm` objects are registered and if not, it automatically activates `UAR(0)` for REPL. Since webREPL only registers a dupterm object when a connection is actually made and we expressly **deactivated** `UART(0)` in `boot.py`, `mp_reset` thus **activates** `UART(0)`...

To me this looks a bit like legacy code/behaviour from before the option of detaching `UART(0)` from REPL was introduced (PR #3784).

So, shall I submit a pull request for this or not? There are two options:

1. `mp_reset` could additionally check whether `dupterm.listen_s is None` before falling back on `UART(0)`.
2. Alternatively it might be acceptable to remove activation of `UART(0)` for REPL entirely from `mp_reset` since once could argue activation of REPL on startup is now handled entirely by `boot.py`. No need to hard code this any more.

dpgeorge, 2018-11-25:
Thanks for the report.  Indeed this is a problem, as described above.  For example, if the user has a device/sensor attached to the UART(0), and wants to drop to a pure webrepl REPL to use this device, then activating the REPL on the UART(0) will most likely mess up the attached sensor.

>  `mp_reset` could additionally check whether `dupterm.listen_s is None` before falling back on `UART(0)`.

That seems way too much of a hack, because `listen_s` is in Python code (webrepl.py) and could easily be changed, breaking the C code.  And this approach doesn't scale to other ways of using dupterm (eg registering another object other than a UART).

>  Alternatively it might be acceptable to remove activation of `UART(0)` for REPL entirely from `mp_reset` since once could argue activation of REPL on startup is now handled entirely by `boot.py`. No need to hard code this any more.

This is too dangerous: if by some accident or otherwise the UART(0) is not activated on the REPL by a user script, and webrepl isn't started (or wifi isn't active) then the device is "bricked" and the only way to recover it is to do a complete erase and reflash of the firmware, with a new default filesystem (or to just upload a fresh filesystem, but that's also not easy to make one).

It seems to me that the only way to resolve this issue is to unconditionally attach UART(0) to dupterm at the very beginning before `boot.py` runs (eg activate it in `_boot.py`), and then the user can choose to explicitly detach it (via `uos.dupterm(None)`) in their `boot.py` or `main.py` (and OS messages would need to be suppressed until the end of executing `boot.py` to prevent spurious characters being sent to anything attached to UART(0)).

dpgeorge, 2019-01-09:
I submitted #4405 as a fix for this issue.  It enables UART(0) as the REPL before running boot.py, so boot.py can disable it if needed.