micropython/micropython #4316: configure.ac:41: error: possibly undefined macro: AC_PROG_LIBTOOL

ARF1, 2018-11-23:
On Debian jessie 64-bit `make deplibs` fails for the unix port, despite all required packages being installed:

```
vagrant@contrib-jessie:/vagrant/micropython/ports/unix$ sudo apt-get install bu
ild-essential libffi-dev pkg-config autoconf
Reading package lists... Done
Building dependency tree
Reading state information... Done
autoconf is already the newest version.
build-essential is already the newest version.
libffi-dev is already the newest version.
pkg-config is already the newest version.
0 upgraded, 0 newly installed, 0 to remove and 51 not upgraded.
vagrant@contrib-jessie:/vagrant/micropython/ports/unix$ make deplibs
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosit
y.
cd ../../lib/libffi; ./autogen.sh
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal --force -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf --force
configure.ac:41: error: possibly undefined macro: AC_PROG_LIBTOOL
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
Makefile:279: recipe for target '../../lib/libffi/configure' failed
make: *** [../../lib/libffi/configure] Error 1
make: *** Deleting file '../../lib/libffi/configure'
vagrant@contrib-jessie:/vagrant/micropython/ports/unix$
```

pfalcon, 2018-11-23:
If you ask for a solution, then (pretty obviously) libtool package is missing.

If you point to the fact that need for libtool should be mentioned in README, then it's true, and please update the title to make it obvious that it's a documentation problem.

ARF1, 2018-11-24:
@pfalcon Thanks. Installing the libtool package helped. I made a pull request to update the documentation.

dpgeorge, 2021-05-18:
Details about the needed packages should be put in a new `ports/unix/README.md` file.

jonnor, 2024-09-29:
There is now decent documentation in how to build in the ports/unix/README.md file. It mentions the need to install libtool

closed 2024-09-29 by jonnor