micropython/micropython #4323: [STM32] ADCAll not working on H743, rtc.wakeup() not working

rolandvs, 2018-11-28:


Meekdai, 2018-12-11:
In the STM official reference manual, I found the temperature of the STM32H743 in the 18 channel of the ADC3. (VBAT is 17 & VREF is 19). Obviously its channel is different from STM32F4
[http://docs.micropython.org/en/latest/library/pyb.ADC.html?highlight=adcall#the-adcall-object](url)

dpgeorge, 2018-12-11:
I can confirm the problems, and there are multiple issues: the H7 uses 16-bit ADC so the formulas for the calibration need to change, the sampling time for the temp sensor must be increased, and the RTC EXTI numbers need to be changed for the H7.  I have some working patches that will be pushed soon.

dpgeorge, 2018-12-12:
The issues here should all be fixed by the following commits: 1b4031ed64bcc8cb207fe7b9527a59d527ada5ab (for RTC wakeup), 0555ada277c5cf84f4dd5dd2dd99b17f101dca78 (for reading of vref and vbat), 6cab8daee0c8ea5d409419781d806daa05c5babe (for reading of temp).

closed 2018-12-12 by dpgeorge