micropython/micropython #4328: [mboot] jumps to itself?

TimV81, 2018-11-30:
https://github.com/micropython/micropython/blob/5f3016c663042657338ab65a20fe8163500c5ba1/ports/stm32/mboot/Makefile#L51

Isn't this supposed to be TEXT1_ADDR ?

dpgeorge, 2018-12-01:
No, it is correct as TEXT0_ADDR.  TEXT0_ADDR should be the location of the vector table for the *application* firmware.  Some boards also have TEXT1_ADDR which is an optional location for the second part of the application firmware.  MBoot always starts at 0x08000000.  See mboot/README.md for usage.

TimV81, 2018-12-04:
Hi Damien, thanks for the quick reply. I'm still trying to wrap my head around this concept though. The mboot program eventually jumps to the APPLICATION_ADDR:

https://github.com/micropython/micropython/blob/5f3016c663042657338ab65a20fe8163500c5ba1/ports/stm32/mboot/main.c#L1195

But doesn't that result in starting itself again if TEXT0 == bootloader address ?

I'd expected it to be a bootloader in a sense that if none of the criteria regarding the firmware update have been met it just starts the main application (being micropython).

I'm trying to get this concept working on a stm32 H7 which has it's main firmware on 0x0802 and the bootloader on 0x0800. But it somehow is not jumping through to 0x0802 (with either TEXT0 or TEXT1).

dpgeorge, 2018-12-04:
> But doesn't that result in starting itself again if TEXT0 == bootloader address ?

TEXT0 address is taken from the board configuration file `boards/BOARDNAME/mpconfigboard.mk` and needs to be configured in that file to be the right address (it can be anything as long as it leaves room for mboot at 0x08000000).

I've just now added support for mboot on PYBv1.x, with details of how to get it working in mboot/README.md, see a1c81761b1c116dfecc31c63414ecf184ba9320b .  Hopefully that can help to get your case working.

TimV81, 2018-12-04:
Thanks! will dive into it.

closed 2018-12-04 by TimV81